espressif/arduino-esp32 #5371: Fixing issue #4820 reverted issue #3460

jgamble-simple, 2021-07-09:
The commit that fixed issue #4820:
560c0f45 Fix dropped SSL connection when buffer gets full. (#4820)

Reverted the commit:
d6b91872 Fix for espressif#3460 issue (#4424)

So now you can replicate issue #3460 again on the master branch

PilnyTomas, 2022-01-27:
Hi, could you please provide some minimal example to reproduce this issue?

jgamble-simple, 2022-01-27:
Yeah.....
 - Do a large upload of any kind using the WifiClientSecure library
 - Disconnect what ever gateway you are using from the internet while the upload is happening (I just unplug my router's ethernet cable)
 - Viola

If you need code let me know....

jgamble-simple, 2022-01-27:
I am using Arduino 2.0.0

PilnyTomas, 2022-01-28:
Thank you for your answer. If you have any code that can be flashed right away, then please do provide it (for example using [pastebin](https://pastebin.com/). As I am working on multiple issues, this will save me some time.

VojtechBartoska, 2022-03-23:
Closing as Expired, if needed please reopen.