espressif/arduino-esp32 #5376: `mkfatfs` entry in `package_esp32_dev_index.json` does not provide an entry for `x86_64-pc-linux-gnu`

msebas, 2021-07-11:
Hello,

I tried to install arduino-esp32 2.0.0-alpha for Sloeber IDE (4.3.3.202004230915) on an openSuSe 15.2 x86_64 (5.3.18-lp152.72-default Kernel) machine. 
I added `https://github.com/espressif/arduino-esp32/releases/download/2.0.0-alpha1/package_esp32_dev_index.json` to the list of third party index urls in "Arduino->Third party index url's" in Preferences . Afterwards I selected the esp32-2.0.0-alpha1 entry in the "Arduino->Platforms and Boards" in Preferences and hit "Apply".
But "Adopting Arduino Board Platform" in step "Installing Arduino Board Plattforms" returned with the following error:
```
Failed to install esp32 
No valid system found for "mkfatfs".
```

I took a look into the `package_esp32_dev_index.json` and found that the entry for `mkfatfs` reads
```
        {
          "version": "0.3.6",
          "name": "mkfatfs",
          "systems": [
            {
              "host": "i686-pc-linux-gnu",
              "url": "https://github.com/lorol/arduino-esp32fatfs-plugin/raw/master/extra/mkfatfs.tar.gz",
              "archiveFileName": "mkfatfs.tar.gz",
              "checksum": "SHA-256:41dc0346a38d6aab5674d4301937868e69471b69d8317ed667d2eeb7c4a077ee",
              "size": "102575"
            },
            {
              "host": "i686-mingw32",
              "url": "https://github.com/lorol/arduino-esp32fatfs-plugin/raw/master/extra/mkfatfs.zip",
              "archiveFileName": "mkfatfs.zip",
              "checksum": "SHA-256:966f378da8bba524197d7d8a919e93b21b4680ead03ae69cac8bb56f8ec864a6",
              "size": "612169"
            }
          ]
        }
```
missing an entry for `x86_64-pc-linux-gnu`.

Is this entry not required, supported or missing out of some other reason?    
If it was simply overlooked please add it.
Sloeber is quite aggressive in terms of configuration strictness. So it might be that it tries to install something not required.

PS: I added it assuming I could reuse the entry for "i686-pc-linux-gnu", but that failed changing the error message to
```
Failed to install esp32
Failed to extract tar.bz2. "/home/Sebastian/programme/eclipse/sloeber/Sloeber/arduinoPlugin/downloads/mkfatfs.tar.gz".
Invalid archive: it must contain a single root folder
```
which is correct, the archive contains a single file called "mkfatfs", instead of the usual folder.

As soon as I have time I am going to provide it with a dummy on my own Webserver to see if this resolves the issue as workaround.

msebas, 2021-07-12:
Hi,

I tried to run the installer with a local version of "mkfatfs.tar.gz" with the structure
```
mkfatfs.tar.gz
|---->mkfatfs
          |--------->mkfatfs
          |--------->package.json
```
adjusted like "mklittlefs.tar.gz".

Not the installation succeeded, but somewhere something is wrong with the settings (`/bin/sh: /bin/xtensa-esp32-elf-g++: Datei oder Verzeichnis nicht gefunden`),  so I am unable to decide if it is working now.
I am trying to fix the settings and then verify if this indeed the solution.

me-no-dev, 2021-07-13:
that error seems to suggest that the toolchain folder was not set properly

msebas, 2021-07-18:
OK. Please excuse the late answer. I was very busy during the week and checking the follow up error was a mess.

My findings:
1. Even with Sloeber 4.4 `mkfatfs` is still required, so one has to use a makeshift version of the `package_esp32_dev_index.json` to circumvent it.
2. The `mkfatfs` binary provided for "i686-pc-linux-gnu" seems to work on "x86_64-pc-linux-gnu", but I have no controller where a predefined FAT partition scheme exists, so I am unable to check if it is really working.

@me-no-dev : I would like to know how we should proceed from here on. I could think of 3 possibilities:
1.) I make merge requests for both this repository and https://github.com/lorol/arduino-esp32fatfs-plugin to include my makeshift changes as parts of the setup config.
2.) One might follow the suggestion at https://github.com/lorol/arduino-esp32fatfs-plugin and drop `mkfatfs` in favor of https://github.com/lorol/arduino-esp32fs-plugin (I cannot see all the implications if this step)
3.) We leave it as it is.
If you have any other suggestion, I am going to help, but I do not have that much time and abilit at my disposal.

Kind regards

PS: For anyone hitting this because googling the follow up erro: 
1. Update to Sloeber 4.4 (If the normal Sloeber does not work, download eclipse C++-Embeeded and install the plugin)
2. Sloeber 4.4 does not support Java 16 which is now bundled to newer Eclipse version, so use the 2021.03 (check that it uses a justj.openjdk 15) and avoid any dependency to bintray by disabling the repos because of https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/

me-no-dev, 2021-07-18:
@msebas thanks for exploring :) I think the proper solution is to remove the tool from the json until the maintainer of it provides versions for all platforms. You do not need it to build and flash through sloeber and espressif is working on own tool that will do more or less the same and be precompiled multi-platform.

closed 2021-07-18 by me-no-dev