espressif/arduino-esp32 #5377: arduino-esp32中人脸识别阈值

dazhao168, 2021-07-12:
在C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\sdk\include\esp-face路径中，fr_forward.h文件内的FACE_REC_THRESHOLD是否已经启用？我将阈值0.55改为0.9或0.1，在识别正确率没有非常明显的改观？
如果要增加正确率需要做那些修改呢？感谢！

我的配置
mtmn_config.min_face = 60;
config.frame_size = FRAMESIZE_QQVGA;

VojtechBartoska, 2021-07-12:
@dazhao168 Please provide more info about your issue by following Issue template. Keep in mind that main language here is English.

dazhao168, 2021-07-14:
@VojtechBartoska Please take a look, can this clearly describe the problem?

VojtechBartoska, 2022-03-30:
Hello @dazhao168  , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

closed 2022-04-22 by VojtechBartoska