espressif/arduino-esp32 #5379: How to use xTaskCreateStatic() and  xTaskCreateStaticPinnedToCore() with Arduino IDE?

DavidHsu100, 2021-07-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                     yes
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
I want to use external RAM(PSRAM) for task. 
I use xTaskCreateStatic() and  xTaskCreateStaticPinnedToCore()  to to hold the new task's data structures (TCB) and the task's stack.
I set the configSUPPORT_STATIC_ALLOCATION flag in hardware\esp32\1.0.4\tools\sdk\include\config\sdkconfig.h
 #define configSUPPORT_STATIC_ALLOCATION 1
But when I compile the code. It has some error message.
My probplem is:
1. What flag is need to set?
2. How the flag to set with Arduino IDE?
3. Which .h file need to include?

### Sketch: 
#include <Arduino.h>
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

// Function that implements the task being created.
void vTaskCode( void * pvParameters )
{
    // The parameter value is expected to be 1 as 1 is passed in the
    // pvParameters value in the call to xTaskCreateStatic().
    configASSERT( ( uint32_t ) pvParameters == 1UL );

    for( ;; )
    {
        // Task code goes here.
    }
}

void setup() {
      StaticTask_t xTaskBuffer;
      TaskHandle_t xHandle = NULL;
      #define STACK_SIZE 1024*10
      EXT_RAM_ATTR StackType_t xStack[ STACK_SIZE ];    
    // Create the task without using any dynamic memory allocation.
    xHandle = xTaskCreateStatic(
                  vTaskCode,       // Function that implements the task.
                  "NAME",          // Text name for the task.
                  STACK_SIZE,      // Stack size in bytes, not words.
                  ( void * ) 1,    // Parameter passed into the task.
                  tskIDLE_PRIORITY,// Priority at which the task is created.
                  xStack,          // Array to use as the task's stack.
                  &xTaskBuffer );  // Variable to hold the task's data structure.
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
C:\Users\xxx\AppData\Local\Temp\arduino_build_442770\sketch\sketch_jul13a.ino.cpp.o:(.literal._Z5setupv+0x20): undefined reference to `xTaskCreateStaticPinnedToCore'

C:\Users\xxx\AppData\Local\Temp\arduino_build_442770\sketch\sketch_jul13a.ino.cpp.o: In function `setup()':

D:\company\Project\eSidebed\coding\arduino\ide\portable\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/freertos/freertos/task.h:615: undefined reference to `xTaskCreateStaticPinnedToCore'

collect2.exe: error: ld returned 1 exit status

exit status 1

lbernstone, 2021-07-13:
CONFIG_* options are compiled into the included IDF libraries.  You cannot change these options without [recompiling the libraries](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md).

DavidHsu100, 2021-07-14:
Dear Sir：

If the path of my project is different from that of esp-idf, how to point the path of build to my project.

Thanks~

atanisoft, 2021-07-22:
@DavidHsu100 The FreeRTOS "Static" creation methods are not enabled in the arduino-esp32 1.0.x releases and unless you are willing to recompile the binaries it will not be usable in those versions. In the 2.0.0 versions of arduino-esp32 these options are enabled by default and no recompilation will be required to use these methods. You can install the 2.0.0-rc1 build as documented [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) by using the "Development release link".

VojtechBartoska, 2022-03-30:
Hello @DavidHsu100, is this still valid? If yes, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-13:
Hello, 

as there was no answer in last 14 days, I'm **closing the issue as expired** to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-13 by VojtechBartoska