espressif/arduino-esp32 #5380: ESP32 Doit Devkit v1 uploading Sketch(s) impossible

ZEROF, 2021-07-13:
### Hardware:
Board:						     ESP32 DOIT Devkit v1	    	
IDE name:					     Tested Arduino IDE (1.8.13 and 2.x beta)
Flash Frequency:				     Tested 40/80Mhz
Upload Speed:				     Tested 921600/115200
Computer OS:                                      Ubuntu 20.04, as well with Ubuntu 18.04/Kernel 5.4.0-77-generic

[image]

### Description:

Before (few months before) this was like riding a bike, connect board to PC, upload what you want, have fun but now it's impossible to upload anything to ESP32, even blink example. I tested with 5 USB cables and 3 ESP32 boards, just to be sure that I don't have some hardware related issue. I tried with holding "BOOT" button as well (never needed to this), but it ends like this:

[image]

If i just try to upload without touching anything, just connecting board to pc (like it was possible before):

[image]

To be sure that board is well connected i try I don't know how many times, and from dmesg I can see (ports ttyUSB0, ttyUSB1, ttyUSB2 etc..)

[image]

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}

```

### Debug Messages:
```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv�<� �W|�KW|�K�r�+��r�+��r�+��r����r�+��r����r����r������K�r�+O,
```

me-no-dev, 2021-07-13:
Your issues are consistent with USB related problems. If not the cable, maybe there is an old esptool process running in the background? Some change in the USB host? Did you try restarting the computer?

theotherjenkutler, 2021-07-14:
I have the exact same problem. I also tried the Sparkfun ESP32 Thing Plus and I have the same problem. When I try uploading code onto a SAMD21 with the same cable and USB port, I have no trouble.

JohnMacrae, 2021-07-14:
I have found all these boards usually fail with automatic uploads. The
design us faulty. Allegedly there is a fix using a large capacitor between
EN and GND. I have not tested it. I usually just add OTA code to my sketch
and use it.

ZEROF, 2021-08-05:
Hi,

I have found solution. ESP32 will not work with USB 3.0 (tested with 5 cables). I picked up my old ThinkPad X201 - yeah +10 years laptop :), when i upload I can see in console "Connecting ...." I push "BOOT" button and sketch upload is working. Arduino IDE or ESP32 have issue with USB 3.0. But I'm not sure if it's related to IDE or ESP32 and who should fix this issues if possible.

lbernstone, 2021-08-05:
esp32 has no USB.  The board you are using has a built-in USB->UART adapter.

VojtechBartoska, 2022-03-30:
Hello, as a part of Issue tracker cleanup I'm closing the issue, if still needed. You can reopen it. Thanks for your contribution.

closed 2022-03-30 by VojtechBartoska