espressif/arduino-esp32 #5381: Problem to upload sketch in esp32-c3-devkitm-1

vanderareosa, 2021-07-13:
I'm trying to upload my sketch.
### Hardware:
I'm using mac 10.11 and arduino IDE 1.8.13
Board ESP32-C3-DevKitM-1 v1.6

### Description:
I always used ESP32 with arduino, but a few weeks ago I bought esp32-c3-devkitm-1 and to my surprise when I tried to upload the sketch, the following error appeared: 

### Sketch: Whatever ..WIFIBasic, Blink .....

### Debug Messages:
esptool.py v3.0-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
WARNING: This chip doesn't appear to be a ESP32 (chip magic value 0x1b31506f). Probably it is unsupported by this version of esptool.
Chip is ESP32-D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core, Coding Scheme None
WARNING: Detected crystal freq 0.00MHz is quite different to normalized freq 26MHz. Unsupported crystal in use?
Crystal is 26MHz
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x47)
A fatal error occurred: Invalid head of packet (0x47)

how can I solve this and go back to using esp32 together with arduino?

lbernstone, 2021-07-13:
esp32-c3 is supported in the 2.0.0alpha version, available as the [development version](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md)

vanderareosa, 2021-07-13:
Thanks. Now my sketch has loaded, but some errors have come up that didn't exist before.

**In file included from /Users/firemedia1/Dropbox/Arduino/libraries/Espalexa/src/Espalexa.h:60,
                 from /Users/firemedia1/Dropbox/Arduino/Plumuf/plumuf_0_9_5/plumuf_0_9_5.ino:13:
/Users/firemedia1/Dropbox/Arduino/libraries/Espalexa/src/EspalexaDevice.h:7:1: warning: 'typedef' was ignored in this declaration
 typedef class EspalexaDevice;
 ^~~~~~~
/Users/firemedia1/Dropbox/Arduino/Plumuf/plumuf_0_9_5/plumuf_0_9_5.ino: In function 'boolean compareDates(String)':
/Users/firemedia1/Dropbox/Arduino/Plumuf/plumuf_0_9_5/plumuf_0_9_5.ino:634:17: warning: NULL used in arithmetic [-Wpointer-arith]
   if ((date != NULL) or (date != "")){
                 ^~~~
/Users/firemedia1/Dropbox/Arduino/Plumuf/plumuf_0_9_5/plumuf_0_9_5.ino: In function 'boolean getFirebaseDBInformations()':
/Users/firemedia1/Dropbox/Arduino/Plumuf/plumuf_0_9_5/plumuf_0_9_5.ino:849:44: warning: control reaches end of non-void function [-Wreturn-type]
    File file = SPIFFS.open(pathFirebirdFile);
                                            ^
/Users/firemedia1/Dropbox/Arduino/libraries/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp: In member function 'IPAddress AsyncWebSocketClient::remoteIP()':
/Users/firemedia1/Dropbox/Arduino/libraries/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp:843:28: error: call of overloaded 'IPAddress(unsigned int)' is ambiguous
         return IPAddress(0U);
                            ^
In file included from /Users/firemedia1/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/Arduino.h:153,
                 from /Users/firemedia1/Dropbox/Arduino/libraries/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp:21:
/Users/firemedia1/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/IPAddress.h:51:5: note: candidate: 'IPAddress::IPAddress(const uint8_t*)'
     IPAddress(const uint8_t *address);
     ^~~~~~~~~
/Users/firemedia1/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/IPAddress.h:50:5: note: candidate: 'IPAddress::IPAddress(uint32_t)'
     IPAddress(uint32_t address);
     ^~~~~~~~~
/Users/firemedia1/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/IPAddress.h:29:7: note: candidate: 'constexpr IPAddress::IPAddress(const IPAddress&)'
 class IPAddress: public Printable**

vanderareosa, 2021-07-13:
Thanks so much!

closed 2021-07-13 by vanderareosa