espressif/arduino-esp32 #5382: Refactoring of UART of ESP-IDF APIs

VojtechBartoska, 2021-07-14:
Related issues:
#5112
#5047 
#5362 

Resolution will be add later.

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			2.0.0 alpha
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

SuGlider, 2021-07-14:
Issue #5112 and issue # 5047 solved and closed with PR https://github.com/espressif/arduino-esp32/pull/5385
Already merged in master

SuGlider, 2021-07-15:
Issue #5362 solved with PR https://github.com/espressif/arduino-esp32/pull/5389

SuGlider, 2021-07-16:
All issues listed are solved/closed.

closed 2021-07-16 by SuGlider