espressif/arduino-esp32 #5386: Getting Windows Error 1073741819 while compilation

blaubaer, 2021-07-15:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.6
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
The whole compilation does not work. There is a strange Windows `1073741819` while compilation. See output here:

```
Arduino: 1.8.15 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 40MHz, 4MB (32Mb), 921600, None"
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\such-a-user\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\such-a-user\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\such-a-user\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=40,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10815 -build-path C:\Users\SUCH~1.USR\AppData\Local\Temp\arduino_build_869902 -warnings=default -build-cache C:\Users\SUCH~1.USR\AppData\Local\Temp\arduino_cache_169878 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\such-a-user\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\such-a-user\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\such-a-user\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=40,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10815 -build-path C:\Users\SUCH~1.USR\AppData\Local\Temp\arduino_build_869902 -warnings=default -build-cache C:\Users\SUCH~1.USR\AppData\Local\Temp\arduino_cache_169878 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
Using board 'esp32' from platform in folder: C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
Using core 'esp32' from platform in folder: C:\Users\such-a-user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
cmd /c if exist "C:\\Program Files (x86)\\Arduino\\examples\\01.Basics\\Blink\\partitions.csv" copy /y "C:\\Program Files (x86)\\Arduino\\examples\\01.Basics\\Blink\\partitions.csv" "C:\\Users\\SUCH~1.USR\\AppData\\Local\\Temp\\arduino_build_869902\\partitions.csv"
exit status -1073741819
Error compiling for board ESP32 Dev Module.
```

### Sketch:

It does not matter which sketch I use. I happens which each sketch. Correct or not.

```cpp
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH); 
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```

### Debug Messages:
Does not matter. I did not even reach there.

VojtechBartoska, 2022-03-30:
Hello @blaubaer, do you still need some help?

VojtechBartoska, 2022-04-13:
Hello, 

as there was no answer in last 14 days, I'm **closing the issue as expired** to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-13 by VojtechBartoska