espressif/arduino-esp32 #5392: ESP32 OTA update  HTTP_CLIENT: Error parse url

krupis, 2021-07-16:
Hello. I have been using the old HTTPUpdate.h library to perform OTA update on my ESP32 devices. From what I know, this library is now deprecated and it is better to use Update library instead.  

My old OTA function:

```
void start_OTA(){

  setup_wifi(); // function to connect to w
  while (WiFi.status() == WL_CONNECTED) {
  WiFiClient client;

       t_httpUpdate_return ret = httpUpdate.update(client, "192.168.4.200",80,"/uploads/PTL.bin");//esp32

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        //Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());//esp8266
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());//esp32
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
    
}
}
```

The code above works fine for me. As you can see, the ESP32 after setting up the WiFi will connect to the Raspberry PI (configured static IP). The raspberry pi have apache webserver and I store the binary in the uploads/esp.bin. The OTA works perfectly fine.

I am now trying to migrate to use Update.h library and I have found the example code here:
https://github.com/espressif/arduino-esp32/tree/master/libraries/Update/examples
Https_ota_update.

I have copied the example code, changed the wifi credentials and raspberry IP, but the OTA does not seem to work:
Full code:
```
// This sketch provide the functionality of OTA Firmware Upgrade
#include "WiFi.h"
#include "HttpsOTAUpdate.h"
// This sketch shows how to implement HTTPS firmware update Over The Air.
// Please provide your WiFi credentials, https URL to the firmware image and the server certificate.

static const char *ssid     = "xxxxxx";  // your network SSID (name of wifi network)
static const char *password = "xxxxxx"; // your network password

static const char *url = "192.168.4.200/uploads/PTL.bin"; //state url of your firmware image

static const char *server_certificate = "-----BEGIN CERTIFICATE-----\n" \
     "MIIEkjCCA3qgAwIBAgIQCgFBQgAAAVOFc2oLheynCDANBgkqhkiG9w0BAQsFADA/\n" \
     "MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
     "DkRTVCBSb290IENBIFgzMB4XDTE2MDMxNzE2NDA0NloXDTIxMDMxNzE2NDA0Nlow\n" \
     "SjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxIzAhBgNVBAMT\n" \
     "GkxldCdzIEVuY3J5cHQgQXV0aG9yaXR5IFgzMIIBIjANBgkqhkiG9w0BAQEFAAOC\n" \
     "AQ8AMIIBCgKCAQEAnNMM8FrlLke3cl03g7NoYzDq1zUmGSXhvb418XCSL7e4S0EF\n" \
     "q6meNQhY7LEqxGiHC6PjdeTm86dicbp5gWAf15Gan/PQeGdxyGkOlZHP/uaZ6WA8\n" \
     "SMx+yk13EiSdRxta67nsHjcAHJyse6cF6s5K671B5TaYucv9bTyWaN8jKkKQDIZ0\n" \
     "Z8h/pZq4UmEUEz9l6YKHy9v6Dlb2honzhT+Xhq+w3Brvaw2VFn3EK6BlspkENnWA\n" \
     "a6xK8xuQSXgvopZPKiAlKQTGdMDQMc2PMTiVFrqoM7hD8bEfwzB/onkxEz0tNvjj\n" \
     "/PIzark5McWvxI0NHWQWM6r6hCm21AvA2H3DkwIDAQABo4IBfTCCAXkwEgYDVR0T\n" \
     "AQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8EBAMCAYYwfwYIKwYBBQUHAQEEczBxMDIG\n" \
     "CCsGAQUFBzABhiZodHRwOi8vaXNyZy50cnVzdGlkLm9jc3AuaWRlbnRydXN0LmNv\n" \
     "bTA7BggrBgEFBQcwAoYvaHR0cDovL2FwcHMuaWRlbnRydXN0LmNvbS9yb290cy9k\n" \
     "c3Ryb290Y2F4My5wN2MwHwYDVR0jBBgwFoAUxKexpHsscfrb4UuQdf/EFWCFiRAw\n" \
     "VAYDVR0gBE0wSzAIBgZngQwBAgEwPwYLKwYBBAGC3xMBAQEwMDAuBggrBgEFBQcC\n" \
     "ARYiaHR0cDovL2Nwcy5yb290LXgxLmxldHNlbmNyeXB0Lm9yZzA8BgNVHR8ENTAz\n" \
     "MDGgL6AthitodHRwOi8vY3JsLmlkZW50cnVzdC5jb20vRFNUUk9PVENBWDNDUkwu\n" \
     "Y3JsMB0GA1UdDgQWBBSoSmpjBH3duubRObemRWXv86jsoTANBgkqhkiG9w0BAQsF\n" \
     "AAOCAQEA3TPXEfNjWDjdGBX7CVW+dla5cEilaUcne8IkCJLxWh9KEik3JHRRHGJo\n" \
     "uM2VcGfl96S8TihRzZvoroed6ti6WqEBmtzw3Wodatg+VyOeph4EYpr/1wXKtx8/\n" \
     "wApIvJSwtmVi4MFU5aMqrSDE6ea73Mj2tcMyo5jMd6jmeWUHK8so/joWUoHOUgwu\n" \
     "X4Po1QYz+3dszkDqMp4fklxBwXRsW10KXzPMTZ+sOPAveyxindmjkW8lGy+QsRlG\n" \
     "PfZ+G6Z6h7mjem0Y+iWlkYcV4PIWL1iwBi8saCbGS5jN2p8M+X+Q7UNKEkROb3N6\n" \
     "KOqkqm57TH2H3eDJAkSnh6/DNFu0Qg==\n" \
     "-----END CERTIFICATE-----";

static HttpsOTAStatus_t otastatus;

void HttpEvent(HttpEvent_t *event)
{
    switch(event->event_id) {
        case HTTP_EVENT_ERROR:
            Serial.println("Http Event Error");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            Serial.println("Http Event On Connected");
            break;
        case HTTP_EVENT_HEADER_SENT:
            Serial.println("Http Event Header Sent");
            break;
        case HTTP_EVENT_ON_HEADER:
            Serial.printf("Http Event On Header, key=%s, value=%s\n", event->header_key, event->header_value);
            break;
        case HTTP_EVENT_ON_DATA:
            break;
        case HTTP_EVENT_ON_FINISH:
            Serial.println("Http Event On Finish");
            break;
        case HTTP_EVENT_DISCONNECTED:
            Serial.println("Http Event Disconnected");
            break;
    }
}

void setup(){

    Serial.begin(115200);
    Serial.print("Attempting to connect to SSID: ");
    WiFi.begin(ssid, password);

    // attempt to connect to Wifi network:
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(1000);
    }

    Serial.print("Connected to ");
    Serial.println(ssid);
    
    HttpsOTA.onHttpEvent(HttpEvent);
    Serial.println("Starting OTA");
    HttpsOTA.begin(url, server_certificate); 

    Serial.println("Please Wait it takes some time ...");
}

void loop(){
    otastatus = HttpsOTA.status();
    if(otastatus == HTTPS_OTA_SUCCESS) { 
        Serial.println("Firmware written successfully. To reboot device, call API ESP.restart() or PUSH restart button on device");
    } else if(otastatus == HTTPS_OTA_FAIL) { 
        Serial.println("Firmware Upgrade Fail");
    }
    delay(1000);
}
```

2 questions: 

1. Do I need to somehow say what port I am trying to connect to? (Port 80). And how do I do that?
2. What is server certificate and how do I use it? Perhaps this is my problem since I did not modify it from the example code.

The serial monitor output:
```
10:38:29.409 -> Attempting to connect to SSID: .Connected to xxxxxxxx
10:38:30.528 -> Starting OTA
10:38:30.528 -> Please Wait it takes some time ...
10:38:30.528 -> E (1261) HTTP_CLIENT: Error parse url 192.168.4.200:80/uploads/PTL.bin
10:38:30.528 -> E (1262) HTTP_CLIENT: Error while setting default configurations
10:38:30.528 -> E (1268) esp_https_ota: Failed to initialise HTTP connection
10:38:31.507 -> Firmware Upgrade Fail
10:38:32.532 -> Firmware Upgrade Fail
10:38:33.514 -> Firmware Upgrade Fail
10:38:34.543 -> Firmware Upgrade Fail
10:38:35.522 -> Firmware Upgrade Fail
10:38:36.504 -> Firmware Upgrade Fail
10:38:37.531 -> Firmware Upgrade Fail

```

me-no-dev, 2021-07-16:
`192.168.4.200/uploads/PTL.bin` is not a valid URL :) `https://192.168.4.200/uploads/PTL.bin` is

krupis, 2021-07-16:
Okay thanks for clarification. Another issue im having is regarding the server certificate. Is there any way to disable that just to see if it works? Is it valid to call begin wihtout the certificate?
HttpsOTA.begin(url,);

krupis, 2021-07-16:
Okay I have followed this guide to create key:

https://stackoverflow.com/questions/19705785/python-3-simple-https-server

I have managed to create my key and then ran this Python script. 

The key generated is just a block of string. Can you help me understand why it is required to have \n new line at the end of each line? Can we not just copy the whole block of string and paste it to *server_certificate?

me-no-dev, 2021-07-16:
that is how keys are stored. New line is also part of the string

me-no-dev, 2021-07-16:
calling `HttpsOTA.begin(url, NULL, true);` should skip the certificate

VojtechBartoska, 2022-03-30:
Hello @krupis, can I consider this as solved?

krupis, 2022-03-31:
Yes this is solved. I am closing the issue. Thanks :)

closed 2022-03-31 by krupis