espressif/arduino-esp32 #5394: Why is the RGB LED so bright?

technoblogy, 2021-07-16:
I've been experimenting with the ESP32 2.0.0-alpha1 core on the Arduino IDE, using the following boards:

* ESP32-C3-DevKitM-1
* ESP32-S2-Saola-1M

In both cases, when you power up the board the RGB LED lights up eye-scorchingly bright. I had to tape a piece of paper over it to make it possible to look at the board without getting spots in front of my eyes. Is this really necessary?

Presumably this can be fixed in the software, but if it's fine for everyone else I'll just put up with it.

SuGlider, 2021-07-16:
RGB LED can be turned off by software and then kept off for any other sketch you may load after it.

One way to do it is to use Adafruit NeoPixel Library.
You will find the instructions and necessary sketch at this link: 
https://www.instructables.com/ESP32-S2-Saola-Making-the-RGB-Work/

In order to turn it off, just change to pixels.setBrightness(0); on setup().
The LED will be kept off for any other sketch you want to load in the future on the S2.

technoblogy, 2021-07-17:
@SuGlider Thank you.

SuGlider, 2021-07-21:
Closing this issue. If necessary, please feel free to reopen it.

closed 2021-07-21 by SuGlider