espressif/arduino-esp32 #5405: (help wanted) Attach pin Interrupt to Core

johnsmakar, 2021-07-17:
### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|TinyPICO|
|<B>Version/Date</B>|IDF release/v3.3
|<B>IDE name</B>|Arduino IDE
|<B>Flash Frequency</B>|80Mhz
|<B>PSRAM enabled</B>|yes
|<B>Upload Speed</B>|921600
|<B>Computer OS</B>|Windows 10

### Description:
Hello, I have a project which I am currently doing multithreading on both cores and would like to attach a hardware interrupt on two pins to core 0. I am am using the Arduino framework and cannot find any examples on how to implement this feature.

I get the following error:
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40091072  PS      : 0x00060e34  A0      : 0x80090267  A1      : 0x3ffbf830  
A2      : 0x3ffb81f0  A3      : 0x3ffdc6a4  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060e23  A7      : 0x00000000  A8      : 0x3ffdc6a4  A9      : 0x3ffdc6a4  
A10     : 0x00000016  A11     : 0x00000016  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060e21  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4008cbdd  LEND    : 0x4008cbed  LCOUNT  : 0xfffffff4  
Core 0 was running in ISR context:
EPC1    : 0x4008f74a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40091072

ELF file SHA256: 0000000000000000

Backtrace: 0x40091072:0x3ffbf830 0x40090264:0x3ffbf850 0x4008e573:0x3ffbf870 0x400dd619:0x3ffbf8b0 
0x400dc1dd:0x3ffbf8d0 0x400dc486:0x3ffbf8f0 0x400dc565:0x3ffbf
```

There is no point to include the code because I am most likely doing it all wrong, is there any example of how it is supposed to be structured?

Thanks!

SuGlider, 2021-07-21:
@johnsmakar 
Just make sure the task that allocated the interrupt is pinned to the core that you want the interrupt to occur on.
The interrupt will always be allocated on the core that runs this function, thus it's always advisable to create interrupt in a task pinned to a specific core.

For more information, please refer to:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/intr_alloc.html

johnsmakar, 2021-07-21:
Hello, thanks so much for the response. I figured it out.

closed 2021-07-21 by johnsmakar