espressif/arduino-esp32 #5409: Support manually changing smartconfig protocol (to new ESPTOUCH_V2)

abobija, 2021-07-17:
ArduinoESP32 currenty uses [hardcoded esptouch V1](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.cpp#L667) protocol for smartconfig, but there is a new improved [version V2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_smartconfig.html#_CPPv418smartconfig_type_t) of esptouch protocol.

It would be great if we were able to manually set the protocol before using `Wifi.beginSmartConfig` method.

I think that would solve #5226 as well.

Wang-Yumin, 2022-02-22:
Is there anyone to help with this problem? Thank you, boss.

me-no-dev, 2022-02-22:
@P-R-O-C-H-Y want to have a go at updating the esp-touch implementation inside WiFi?

closed 2022-03-02 by me-no-dev

Wang-Yumin, 2022-03-02:
@P-R-O-C-H-Y Great! Thank you so much!

abobija, 2022-03-02:
Thanks @P-R-O-C-H-Y 👍🏼

PR #6367 (2022-03-02): FEATURE: Implemented new types of Wifi SmartConfig
## Summary
Implemented all types of Wifi SmartConfig. 

Function changed from:
`bool beginSmartConfig();`
to
`bool beginSmartConfig(smartconfig_type_t type = SC_TYPE_ESPTOUCH, char* crypt_key = NULL);`

If `WiFi.beginSmartConfig();` is called, ESPTOUCH is used as default type.
Or you can specify type nad crypt key (used for ESPTOUCH_V2 only):
```
char crypt_key[] = "ESPWIFISMARTCONF"; //should be 16 chars lenght
WiFi.beginSmartConfig(SC_TYPE_ESPTOUCH_V2, crypt_key);
```

| SmartConfig types | Protocol used |
| ------------- | ------------- |
| [... cut]