espressif/arduino-esp32 #5411: [documentation] std::thread: is it legal and failsafe to use int main() and redefine the default  setup() and loop() ?

dsyleixa, 2021-07-18:
std::thread: is it legal and failsafe to use int main() and redefine setup() and loop() ?
eg., 

```
std::thread *thread_1;
std::thread *thread_2;
std::thread *thread_3;
std::thread *thread_4;

void f1() { }

void f2() { }

void f3() { }

void f4() { }

void setup() {
  Serial.begin(115200);
  thread_1 = new std::thread(f1);
  thread_2 = new std::thread(f2);
  thread_3 = new std::thread(f3);
  thread_4 = new std::thread(f4);
}

void loop() {    
    static uint32_t main_loop_counter = 0;
    Serial.println((String)"\nmain loop: " + main_loop_counter);
    delay(500);
    main_loop_counter++;
}

void main() {
   setup();
   while(true){ loop(); }
}

```

in my tests it worked, not sure if this is RTOS-compliant though

igrr, 2021-07-18:
Defining `void main()` is legal, however your `main` function won't actually be called.

Your `setup` and `loop` function will be invoked from `loopTask` defined here: https://github.com/espressif/arduino-esp32/blob/0aafa05e8f4ab51a08d405c53dcb5ef8ef21f15e/cores/esp32/main.cpp#L33-L46.

The way `setup` and `loop` are invoked has little influence on threads created using `std::thread`, though. That part of your sketch looks valid.

dsyleixa, 2021-07-18:
thanks you!
my int main() of course was stripped down, it might also have an infinitive internal loop inside.

I just was reworking my question as I also was unsure about joining threads:

(2) is thread.join() available/needed?

// *SNIP*
thread_1 = new std::thread(f1);
thread_2 = new std::thread(f2);
thread_3 = new std::thread(f3);
thread_4 = new std::thread(f4);

std::thread_1.join();
std::thread_2.join();
std::thread_3.join();
std::thread_4.join();

dsyleixa, 2021-07-18:
thank you!
my int main() of course was stripped down, it might also have more extra code inside.

I just was reworking my question as I also was unsure about joining threads:

(2) is thread.join() available/needed?

// SNIP
thread_1 = new std::thread(f1);
thread_2 = new std::thread(f2);
thread_3 = new std::thread(f3);
thread_4 = new std::thread(f4);

std::thread_1.join();
std::thread_2.join();
std::thread_3.join();
std::thread_4.join();

igrr, 2021-07-18:
> (2) is thread.join() available/needed?

This depends on what you want to achieve. If you want to wait at some point in your program until each thread has finished, you do need to call `join`.

dsyleixa, 2021-07-18:
yes, sort of that is what I want to do. But coding (like for Unix C++ std::thread cppreference)
std::thread_1.join();
std::thread_2.join();
std::thread_3.join();
std::thread_4.join();
I get the error:

```
exit status 1
'thread_1' is not a member of 'std'
```

which is the correct syntax for RTOS?

dsyleixa, 2021-07-18:
ah - I thin I got it:

      thread_1->join();
      thread_2->join();
      thread_3->join();
      thread_4->join();

igrr, 2021-07-18:
`std::` is a namespace, and `thread` is a class defined in that namespace. When you declare a variable of type `std::thread`, you don't need to prefix the variable name with `std::`.

Since you mention cppreference, please take a look at the example on this page: https://en.cppreference.com/w/cpp/thread/thread/thread.
You need something like: https://coliru.stacked-crooked.com/a/9f3fe8d1de4db13e

closed 2021-07-18 by igrr

dsyleixa, 2021-07-18:
yes, my mistake before was using
`  thread_1.join();` (C++ std:: cppreference)

instead of
`  thread_1->join();` (ESP32 RTOS)

dsyleixa, 2021-07-18:
obviously all threads never can be joined because loop() never can be terminated:
loop() runs forever, so either join is discarded
(no message ever by  Serial.println((String)"\nall threads joined\n"); )

```
// std::thread for ESP32, Arduino IDE

// ver 0.0.3a fibonacci, GPIO, blink, sort und main counter

#include <Arduino.h>
#include <thread>
#include <freertos/task.h>

#ifndef LED_BUILTIN
#define LED_BUILTIN 13
#endif

volatile static bool THREADRUN = true;

uint32_t fibonacci(int n) {
   if(n == 0) {
      return 0;
   } else if(n == 1) {
      return 1;
   } else {
      return (fibonacci(n-1) + fibonacci(n-2));
   }
}

void shellsort(int size, int* A)
{
   int i, j, increment;
   int temp;
   increment = size / 2;

   while (increment > 0) {
      for (i = increment; i < size; i++) {
         j = i;
         temp = A[i];
         while ((j >= increment) && (A[j-increment] > temp)) {
            A[j] = A[j - increment];
            j = j - increment;
         }
         A[j] = temp;
      }

      if (increment == 2)
         increment = 1;
      else
         increment = (unsigned int) (increment / 2.2);
   }
}

#define tpin1  11  // GPIO test pins digitalWrite
#define tpin2  12  // GPIO test pins digitalWrite
#define tpin3  10  // GPIO test pins digitalRead

int32_t test_GPIO() {   //
   volatile static bool w=false, r;

   uint32_t y;
   for (y=0; y<100000; y++) {
      digitalWrite(tpin1, w);
      w=!w;
      r=digitalRead(tpin3);
      digitalWrite(tpin2, w&!r);
   }
   return 1;
}

void blinker_loop() {
   thread_local uint32_t counter = 0;
   vTaskPrioritySet(NULL,0); //set Priority
   Serial.println((String)"blinker_loop Current priority :" + uxTaskPriorityGet(NULL)+"\n");
   while(THREADRUN) {
      digitalWrite(LED_BUILTIN, HIGH);
      Serial.println((String)"\nblinker_loop (HIGH) counter: "+ counter+"\n");
      //std::this_thread::sleep_for(std::chrono::milliseconds(1000));
      delay(1000);

      digitalWrite(LED_BUILTIN, LOW);
      Serial.println((String)"\nblinker_loop (LOW) counter: "+ counter+"\n");
      //std::this_thread::sleep_for(std::chrono::milliseconds(500));
      delay(500);

      counter++;
      if(!THREADRUN) {
         Serial.println((String)"blink_loop terminated by THREADRUN");
      }
   }
}

void fibonacci_loop() {
   thread_local uint32_t counter = 0, i=0;
   vTaskPrioritySet(NULL,0);//set Priority
   Serial.println((String)"fibonacci_loop Current priority :" + uxTaskPriorityGet(NULL)+"\n");
   while(THREADRUN) {
      for(i=25; i<41; i++) {    // limits: test, debug
         if(!THREADRUN) {
            Serial.println((String)"fibonacci_loop terminated by THREADRUN");
            break;
         }
         Serial.println( (String)"\nfibonacci of "+i+"="+fibonacci(i)+"\n");
      }
      counter++;
      Serial.println((String)"\nfibonacci_loop counter: "+counter+"\n");
   }
}

void sort_loop() {
   static int array[1000];

   thread_local uint32_t counter = 0, i=0;
   vTaskPrioritySet(NULL,0);//set Priority
   Serial.println((String)"sort_loop Current priority :" + uxTaskPriorityGet(NULL)+"\n");
   while(THREADRUN) {
      Serial.println((String)"\nsort_loop counter: "+counter+"\n");

      for(i=0; i<1000; i++) {
         array[i]= random(0,65000);
      }
      shellsort(1000, array);
      for(i=0; i<1000; i+=50) { // abgekürzte Ausgabe  da Serial sehr langsam
         Serial.println((String)i + "="+array[i]+"\n");
      }
      counter++;
      if(!THREADRUN) {
         Serial.println((String)"sort_loop terminated by THREADRUN");
      }
   }
}

void GPIO_loop() {

   thread_local uint32_t counter = 0, i=0, timerms;
   vTaskPrioritySet(NULL,0);//set Priority
   Serial.println((String)"GPIO_loop Current priority :" + uxTaskPriorityGet(NULL)+"\n");
   while(THREADRUN) {
      Serial.println((String)"\nGPIO_loop counter: "+counter);
      timerms = millis();
      test_GPIO();
      timerms = millis() - timerms;
      Serial.println((String)"\nGPIO test timerms=" + (String)timerms + "\n");
      counter++;
      if(!THREADRUN) {
         Serial.println((String)"GPIO_loop terminated by THREADRUN");
      }
   }
}

std::thread thread_1 (blinker_loop);
std::thread thread_2 (fibonacci_loop);
std::thread thread_3 (sort_loop);
std::thread thread_4 (GPIO_loop);

void setup() {
   Serial.begin(115200);
   delay(1000);

   pinMode(LED_BUILTIN, OUTPUT);
   pinMode(tpin1, OUTPUT);
   pinMode(tpin2, OUTPUT);
   pinMode(tpin3, INPUT_PULLUP);

}

volatile static uint32_t loop_counter = 0;

void loop() {
   Serial.println((String)"\nloop counter: " + loop_counter);
   delay(500);
   loop_counter++;
   if(loop_counter>40) {
      Serial.println("loop counter limit reached!  all threads terminate!");
      THREADRUN=false; // after 20sec
   }
}

int  main() {
   setup();
   while(THREADRUN) {
      loop();                 // should terminate when THREADRUN==false, but fails!
      if(!THREADRUN) break;   // optionally, fails too!
   }

   thread_1.join();
   thread_2.join();
   thread_3.join();
   thread_4.join();

   Serial.println((String)"\nall threads joined\n");

   return 0;
}
```

dsyleixa, 2021-07-18:
update:
 not possible to join threads in main, it has to be done in setup.
int main() has to be dropped.
The standard loop() must be empty,
new extra_loop() thread needed to generate terminate signal.

```

// std::thread for ESP32, Arduino IDE

// ver 0.0.3a fibonacci, GPIO, blink, sort und main counter

#include <Arduino.h>
#include <thread>
#include <freertos/task.h>

#ifndef LED_BUILTIN
#define LED_BUILTIN 13
#endif

volatile static bool THREADRUN = true;

uint32_t fibonacci(int n) {
   if(n == 0) {
      return 0;
   } else if(n == 1) {
      return 1;
   } else {
      return (fibonacci(n-1) + fibonacci(n-2));
   }
}

void shellsort(int size, int* A)
{
   int i, j, increment;
   int temp;
   increment = size / 2;

   while (increment > 0) {
      for (i = increment; i < size; i++) {
         j = i;
         temp = A[i];
         while ((j >= increment) && (A[j-increment] > temp)) {
            A[j] = A[j - increment];
            j = j - increment;
         }
         A[j] = temp;
      }

      if (increment == 2)
         increment = 1;
      else
         increment = (unsigned int) (increment / 2.2);
   }
}

//--------------------------------------------------------------------
#define tpin1  11  // GPIO test pins digitalWrite
#define tpin2  12  // GPIO test pins digitalWrite
#define tpin3  10  // GPIO test pins digitalRead

int32_t test_GPIO() {   //
   volatile static bool w=false, r;

   uint32_t y;
   for (y=0; y<100000; y++) {
      digitalWrite(tpin1, w);
      w=!w;
      r=digitalRead(tpin3);
      digitalWrite(tpin2, w&!r);
   }
   return 1;
}

//--------------------------------------------------------------------
void blinker_loop() {
   thread_local uint32_t counter = 0;
   vTaskPrioritySet(NULL,0); //set Priority
   Serial.println((String)"blinker_loop Current priority :" + uxTaskPriorityGet(NULL)+"\n");
   while(THREADRUN) {
      digitalWrite(LED_BUILTIN, HIGH);
      Serial.println((String)"\nblinker_loop (HIGH) counter: "+ counter+"\n");
      //std::this_thread::sleep_for(std::chrono::milliseconds(1000));
      delay(1000);

      digitalWrite(LED_BUILTIN, LOW);
      Serial.println((String)"\nblinker_loop (LOW) counter: "+ counter+"\n");
      //std::this_thread::sleep_for(std::chrono::milliseconds(500));
      delay(500);

      counter++;
      if(!THREADRUN) {
         Serial.println((String)"blink_loop terminated by THREADRUN");
      }
   }
}

//--------------------------------------------------------------------
void fibonacci_loop() {
   thread_local uint32_t counter = 0, i=0;
   vTaskPrioritySet(NULL,0);//set Priority
   Serial.println((String)"fibonacci_loop Current priority :" + uxTaskPriorityGet(NULL)+"\n");
   while(THREADRUN) {
      for(i=25; i<41; i++) {    // limits: test, debug
         if(!THREADRUN) {
            Serial.println((String)"fibonacci_loop terminated by THREADRUN");
            break;
         }
         Serial.println( (String)"\nfibonacci of "+i+"="+fibonacci(i)+"\n");
      }
      counter++;
      Serial.println((String)"\nfibonacci_loop counter: "+counter+"\n");
   }
}

//--------------------------------------------------------------------
void sort_loop() {
   static int array[1000];

   thread_local uint32_t counter = 0, i=0;
   vTaskPrioritySet(NULL,0);//set Priority
   Serial.println((String)"sort_loop Current priority :" + uxTaskPriorityGet(NULL)+"\n");
   while(THREADRUN) {
      Serial.println((String)"\nsort_loop counter: "+counter+"\n");

      for(i=0; i<1000; i++) {
         array[i]= random(0,65000);
      }
      shellsort(1000, array);
      for(i=0; i<1000; i+=50) { // abgekürzte Ausgabe  da Serial sehr langsam
         Serial.println((String)i + "="+array[i]+"\n");
      }
      counter++;
      if(!THREADRUN) {
         Serial.println((String)"sort_loop terminated by THREADRUN");
      }
   }
}

//--------------------------------------------------------------------
void GPIO_loop() {

   thread_local uint32_t counter = 0, i=0, timerms;
   vTaskPrioritySet(NULL,0);//set Priority
   Serial.println((String)"GPIO_loop Current priority :" + uxTaskPriorityGet(NULL)+"\n");
   while(THREADRUN) {
      Serial.println((String)"\nGPIO_loop counter: "+counter);
      timerms = millis();
      test_GPIO();
      timerms = millis() - timerms;
      Serial.println((String)"\nGPIO test timerms=" + (String)timerms + "\n");
      counter++;
      if(!THREADRUN) {
         Serial.println((String)"GPIO_loop terminated by THREADRUN");
      }
   }
}

//--------------------------------------------------------------------

void extra_loop() {
   volatile static uint32_t loop_counter = 0;
   vTaskPrioritySet(NULL,0);//set Priority
   
   while(THREADRUN) {      
      Serial.println((String)"\nloop counter: " + loop_counter);
      delay(500);
      loop_counter++;
      if(loop_counter>40) {
         Serial.println("loop counter limit reached!  all threads terminate!");
         THREADRUN=false; // after 20sec
      }
   }
}

//--------------------------------------------------------------------
//--------------------------------------------------------------------
void setup() {
   Serial.begin(115200);
   delay(1000);

   pinMode(LED_BUILTIN, OUTPUT);
   pinMode(tpin1, OUTPUT);
   pinMode(tpin2, OUTPUT);
   pinMode(tpin3, INPUT_PULLUP);

   std::thread thread_0 (extra_loop);
   std::thread thread_1 (blinker_loop);
   std::thread thread_2 (fibonacci_loop);
   std::thread thread_3 (sort_loop);
   std::thread thread_4 (GPIO_loop);

   thread_0.join();
   thread_1.join();
   thread_2.join();
   thread_3.join();
   thread_4.join();

   Serial.println((String)"\nall threads joined, program terminated\n");  // <<<<<< NEVER REACHED!

}

//--------------------------------------------------------------------
void loop() {
}

```