espressif/arduino-esp32 #5414: [documentation] std::thread: multiple threads executed on both ESP32 cores?

dsyleixa, 2021-07-18:
for` std::thread`: are multiple threads automatically executed on both ESP32 cores?
Or can I choose which threads will have to run on which core exclusively?

atanisoft, 2021-07-19:
std::thread internally will use pthread to create the threads, the default configuration (on master/2.0.0-alpha1) is:
https://github.com/espressif/arduino-esp32/blob/4a0305a05e195b9e8c338c9c07cdb6eaf3dde2df/tools/sdk/esp32/sdkconfig#L1329-L1336. These defaults can be overridden by your code as seen in [this](https://github.com/espressif/esp-idf/tree/master/examples/cxx/pthread) example (which will only work for arduino-esp32 2.0.0-alpha1 or later)

For 1.0.6 it is similarly configured, but you have no ability to override these defaults.

dsyleixa, 2021-07-20:
thanks, unfortunately I don't understand what's happening there.
So given I have 3 to  6 different std::threads, how and where will they run on which cores?

dsyleixa, 2021-07-20:
thanks, unfortunately I don't understand what's happening there.
So given I have 3 to 6 different std::threads, how and where will they run on which cpu cores?
(I am working with ESP32 driver  version 1.06 (latest available in Arduino IDE), and partially 1.04).

atanisoft, 2021-07-20:
> how and where will they run on which cpu cores?

They will run wherever the scheduler puts them, you have no control over it. You also have no control over priorities (they all run at priority 5). A std::thread *WILL* block the loopTask (where setup() and loop() run from) so keep this priority issue in mind.

Unfortunately, std::thread seems rather incomplete since it does not specify how thread priority, stack, etc, should be handled. There is no exposed method for controlling any of these as part of arduino-esp32 1.0.6, in arduino-esp32 2.0.0-alpha1 (or later) there is a method to configure the running core, stack size and priority but it is not through the std::thread API (which intentionally excludes these details).

dsyleixa, 2021-07-20:
thanks, but strange though, I am using
vTaskPrioritySet(NULL,0); //set Priority
in either std::thread which seems to work better if there are extra main loops  (perhaps otherwise blocking wthout this command).

But as to the cpu cores it#s fine then if both cores are used.

atanisoft, 2021-07-20:
> I am using
> vTaskPrioritySet(NULL,0); //set Priority
> in either std::thread which seems to work better if there are extra main level loops

This may work but you are setting the priority to the lowest possible value which is lower than the loopTask (setup/loop executor) so you may unintentionally block that as well. I'd suggest swap the 0 (zero) for ESP_TASK_MAIN_PRIO (constant defined in esp_task.h)

dsyleixa, 2021-07-20:
yes, thank you, I'll immediately  try that too!
vTaskPrioritySet( NULL, ESP_TASK_MAIN_PRIO ); // (correct?)

atanisoft, 2021-07-20:
That looks fine

closed 2021-07-20 by dsyleixa