espressif/arduino-esp32 #5420: ESP Rainmaker Crashing Bug - OTA

JohnMacrae, 2021-07-19:
What you are trying to do?
OTA update via Dashboard with Arduino IDE-compiled binary.

Describe your system( Hardware, computer, O/S, core version, environment).
Ubuntu 20.04, Dell XPS15 9550, Arduino IDE 1.8.15, Latest arduino-esp32 2.0.0a

Describe what is failing.
Replication instruction:
Take Rainmaker Switch example and change Line 65 to:     RMaker.enableOTA(OTA_USING_TOPICS);
Compile, upload and confirm normal behaviour.
Now upload same binary to dashboard with random version number.  Create OTA job and force update - watch serial output as you press the button. Continuous restarts.

Backtrace:0x400d7b93:0x3ffce8400x40177825:0x3ffce870 0x40177ac4:0x3ffce8a0 
Decoding stack results
0x400d7b93: loopTask(void*) at /home/john/arduino-1.8.15/hardware/espressif/esp32/cores/esp32/main.cpp line 35
0x40177825: esp_rmaker_handle_work_queue at ../components/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_core.c line 354
0x40177ac4: esp_rmaker_task at ../components/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_core.c line 445

Now go to dashboard and cancel the OTA job. Watch serial: device reverts to normal behaviour.

[RM Serial output.txt](https://github.com/espressif/arduino-esp32/files/6844034/RM.Serial.output.txt)

JohnMacrae, 2021-11-21:
This behaviour also confirmed with CLI

VojtechBartoska, 2022-04-13:
Hello, 

as there was no answer in last 14 days, I'm **closing the issue as expired** to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

VojtechBartoska, 2022-07-14:
@JohnMacrae I'm reopening the issue.

me-chiel, 2022-08-16:
Same here. When trying the OTA from the dashboard, and OTA in the Arduino IDE it reboots after connecting with WiFi.
(when using RMaker.enableOTA(OTA_USING_TOPICS); ) help would be appreciated. (ESP2.0.4 board definitions installed.)

PR #7242 (2022-09-14): Fixed issue of device crashing in Rainmaker OTA  #5420
Fixed issue wherein device would crash after during OTA upgrade.
I've tested this using ESP32 on Arduino.
Closes #5420 and espressif/esp-rainmaker#161
closed 2022-09-13 by me-no-dev