espressif/arduino-esp32 #5423: dd25e2b breaks some integer to string conversion functions, for example ultoa.

rtrbt, 2021-07-20:
### Description:
The fix for #5045 in commit dd25e2b tries to fix an issue with multiple definitions for reverse in stdlib_noniso.c and esp_dsp.h. However this breaks some integer to string conversion functions, that use the reverse function, at least ultoa, as well as the String constructors using those functions.

Expected output is 123456 (twice), but 654321 is printed instead.

If I change the guard in line 33 to #if 1 it works as expected. https://github.com/espressif/arduino-esp32/blob/4ada3f5804e8165680141653377826d30a4ca364/cores/esp32/stdlib_noniso.c#L33-L47

### Sketch:
```cpp
void setup() {
  // Wait for serial monitor to attach
  delay(3000);
  Serial.begin(115200);

  // Using string constructor
  String s = String(123456ul, 10);
  Serial.println(s);

  // Using ultoa directly
  char buf[10] = {0};
  ultoa(123456ul, buf, 10);
  Serial.println(buf);
}

void loop() {}
```

me-no-dev, 2021-07-20:
In that case, maybe changing it to static could do the trick. Could you please try to remove the guards and rename the function to `_reverse`, then use the new name throughout stdlib_noniso?

rtrbt, 2021-07-20:
Both versions (`static void reverse` and `void _reverse`) seem to work, but I don't have checked if one of the variants collides with definitions in esp_dsp.h. However I would not expect that esp_dsp.h is pulled in by esp_system.h (as it seems to be a separate library?) so marking the function static should be the way to go.

VojtechBartoska, 2022-03-30:
Hello, I'm closing this issue as there is already merged PR. Thanks for your contribution @rtrbt!

closed 2022-03-30 by VojtechBartoska