espressif/arduino-esp32 #5424: Unable to compile Arduino-esp32 with latest ESP-IDF version (4.2 with esp2020r3 toolchain)

GKFernandes, 2021-07-20:
### Hardware:
Board: ESP32-Devkitc
Core Installation version: arduino-esp32 v2.0.0-alpha1
IDE name: IDF component
PSRAM enabled: No
Upload Speed: 115200
Computer OS: Windows 10

### Description:

I'm trying to compile Arduino-esp32 as an ESP-IDF component, I've tried v1.0.5, v1.0.6 and v.2.0.0-alpha1 so far and did not get it work, it seems like it is some kind of mismatch between the ESP-IDF and the Arduino-esp32 versions, because the error is related to some kind of undeclared variable used in a function.

The steps to reproduce my errors are:

Download the amazon-freertos latest repo at: https://github.com/aws/amazon-freertos, used version 202107.00

Install the ESP-IDF newest toolchain, by going to the amazon-freertos/vendors/espressif/esp-idf folder and running install.bat, this installs the esp-2020r3-8.4.0 toolchain.

Then, run the export.bat to add the toolchain to PATH, the following cmd print shows the toolchain being added:

[image]

After that, go to amazon/freertos/vendors/espressif/components and, place the arduino component inside the components folder, like the instruction over here: https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

Then go to the amazon-freertos folder and run the cmake command to generate the build files: cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -S . -B build -GNinja

The build files are generated without any problem and I can run idf.py menuconfig with no issues.

What happens is when I run idf.py build, the following shows up at the very end of the build:

[log_compilation_arduino_esp32.txt](https://github.com/espressif/arduino-esp32/files/6848314/log_compilation_arduino_esp32.txt)

I did not make any changes to the aws demo sketch, I'm trying to compile it as-is from the repo.

Did anyone manage to compile something similar? If it is a mismatch between the arduino component and the ESP-IDF versions, what would it take for it to work?

Thanks in advance.

me-no-dev, 2021-07-20:
arduino master currently compiles with esp-idf master (4.4)

closed 2021-07-20 by me-no-dev

GKFernandes, 2021-07-20:
Hi @me-no-dev ,

The issue was not about the esp-idf master, but the amazon-freertos one.

I hope you can take a look at it.

Thanks,