espressif/arduino-esp32 #5427: Arduino-ESP32 2.0.0-rc1 not compatible with ESPAsyncWebServer

szerwi, 2021-07-20:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			2.0.0-rc1
IDE name:						Arduino IDE 2.0.0

### Description:
2.0.0-rc1 version of Arduino-ESP32 is not compatible with ESPAsyncWebServer library (https://github.com/me-no-dev/ESPAsyncWebServer).
This is the error that I get during compilation:
```
.../arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r1/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: ...\Temp\arduino-sketch-8B419630481664C511ED63E398298E50\libraries\ESPAsyncWebServer\WebAuthentication.cpp.o: in function `getMD5(unsigned char*, unsigned short, char*)':
...\Arduino\libraries\ESPAsyncWebServer\src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'
```

agners, 2021-07-22:
Related: https://github.com/me-no-dev/ESPAsyncWebServer/pull/999

szerwi, 2021-07-29:
Fixed by https://github.com/me-no-dev/ESPAsyncWebServer/pull/999

closed 2021-07-29 by szerwi