espressif/arduino-esp32 #5428: Re: SimpleWiFiServer example; here: submit button click states for HIGH and LOW to different website instances

dsyleixa, 2021-07-20:
as to wifiserver example: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino

after having mouse a "buttonlink" mouse-clicked for LED HIGH or LOW, either a /H or a /L is added to the url, e.g.
http://mysite.net/
becomes 
http://mysite.net/H
which then remains unchanged until a new mouse click.

Given I have opened the website on 2 different devices (PC, smartphone)  and having enabed autorefresh (10s), how can I submit a mousclick (e.g., to HIGH) on the 1st device to the website on the 2nd device (to change the former LOW over there also to HIGH and vice versa?
Otherwise either device will always send the local state, so the 1st website will always send 
http://mysite.net/H for HIGH,  
whilst the 2nd device would repeatedly  send http://mysite.net/L for reset to LOW after each reset and the 1st one then back to HIGH and so on, so it always will toggle to infinity.

(not sure If could describe and express the problem correctly, tbh)

dsyleixa, 2021-09-20:
still not answered or even solved

me-no-dev, 2021-09-20:
i do not see how this is in any way connected to Arduino. The basic example is not meant to do what you want it to do. Surely you can look into HTML and JavaScript and try to figure out how to do it. Refreshing the same URL is not the answer

closed 2021-09-20 by me-no-dev