espressif/arduino-esp32 #5431: Board list request: LOLIN D32 Pro

AlyoshaVasilieva, 2021-07-20:
Per [Espressif board docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/boards/boards.html#third-party) page:

>All the information must be provided by the vendor. If your favorite board is not here, consider creating an issue on GitHub and directly link/mention the vendor in the issue description.

Board: [LOLIN D32 Pro](https://www.wemos.cc/en/latest/d32/d32_pro.html)
Vendor: @wemos (company now branded as LOLIN)

VojtechBartoska, 2021-07-21:
@pedrominatel we can provide link to Wemos D32 docs.

pedrominatel, 2021-07-21:
Hi @wemos. Can you add details about the boards in the official documentation?

We can add links but we need to guarantee that all information provided is the latest and the vendor has agreed with it.

wemos, 2021-08-14:
@pedrominatel 
LOLIN D32 Pro
https://www.wemos.cc/en/latest/d32/d32_pro.html

wemos, 2021-08-14:
LOLIN D32
https://www.wemos.cc/en/latest/d32/d32.html

pedrominatel, 2021-08-16:
@VojtechBartoska and @wemos PTAL at the linked PR.

closed 2021-08-18 by pedrominatel