espressif/arduino-esp32 #5432: Ble start & stop , cause memory

savagerex, 2021-07-21:
I use Ble start & stop . change many times in core 1.0.4.
the heap(memory) will decrease. after 15 times, esp32 will reboot.

i know in core 2.0.0. there is no this problem.

the the heap(memory) still decrease. until memory to 50k. the memory stop decrease.

we hope this problem in 1.0.4 will be solve.

because we need https communication. when https communication. it need memory.

so memory will not enough.

Hardware:
Hardware:
Board : ESP32 dev module
Core install version: 2.0.0
IDE name : Arduino
Flash Frequency : 80MHz
PSRAM Enable : yes
upload speed :115200
computer os : win10

### Description:
Describe your problem here

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

savagerex, 2021-07-21:
the code as below

#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

#define BUTTON_BLE        34
int sensorValueblewifi = 0;
BLEServer *pServer = NULL;
BLECharacteristic * pTxCharacteristic;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint8_t txValue = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();

      if (rxValue.length() > 0) {
        Serial.println("*********");
        Serial.print("Received Value: ");
        for (int i = 0; i < rxValue.length(); i++)
          Serial.print(rxValue[i]);

        Serial.println();
        Serial.println("*********");
      }
    }
};

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("UART Service");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
                    CHARACTERISTIC_UUID_TX,
                    BLECharacteristic::PROPERTY_NOTIFY
                  );
                      
  pTxCharacteristic->addDescriptor(new BLE2902());

  BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
                       CHARACTERISTIC_UUID_RX,
                      BLECharacteristic::PROPERTY_WRITE
                    );

  pRxCharacteristic->setCallbacks(new MyCallbacks());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");

}

void BleUartSetup() {
  //Serial.begin(115200);
  Serial.println("[BleUart] MethodCall BleUartSetup");

  // Create the BLE Device
  BLEDevice::init("UART Service");
  //BLEDevice::setMTU(mtu);

  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
                    CHARACTERISTIC_UUID_TX,
                    BLECharacteristic::PROPERTY_NOTIFY
                  );

  pTxCharacteristic->addDescriptor(new BLE2902());

  BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
                       CHARACTERISTIC_UUID_RX,
                      BLECharacteristic::PROPERTY_WRITE
                    );

  pRxCharacteristic->setCallbacks(new MyCallbacks());

  // Start the service
  pService->start();

  // Start advertising
  // pServer->getAdvertising()->start(); // for backward compability
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.printf("[BleUart] UART Service started(BLE address: %s), waiting a client connection to notify ...\n", BLEDevice::getAddress().toString().c_str());
}

void BleStart()
{
    BleUartSetup();
}

void BleStop()
{
   BLEDevice::deinit(false);
}

void DumpStatistics() {
  Serial.printf("[Psram] Free:%i, Size:%i, MinFree:%i, MaxAlloc:%i\n", ESP.getFreePsram(),ESP.getPsramSize(), ESP.getMinFreePsram(), ESP.getMaxAllocPsram());
  Serial.printf("[Heap] Free:%i, Size:%i, MinFree:%i, MaxAlloc:%i\n", ESP.getFreeHeap(), ESP.getHeapSize(), ESP.getMinFreeHeap(), ESP.getMaxAllocHeap());
  Serial.printf("[Stack] TaskGetStackHighWaterMark: %i\n", uxTaskGetStackHighWaterMark(NULL));
}

void loop() {
  // put your main code here, to run repeatedly:
  sensorValueblewifi = analogRead(BUTTON_BLE);

  //Serial.println(sensorValueblewifi);
  delay(50);

  if (2715 < sensorValueblewifi && sensorValueblewifi < 3374)
  {
      Serial.println("Start");
      BleStart();
      //DumpStatistics();
      Serial.printf("[Heap] Free:%i, Size:%i, MinFree:%i, MaxAlloc:%i\n", ESP.getFreeHeap(), ESP.getHeapSize(), ESP.getMinFreeHeap(), ESP.getMaxAllocHeap());
  }

  if (1430 < sensorValueblewifi && sensorValueblewifi < 1744)
  {
      Serial.println("Stop");
      BleStop();
      //DumpStatistics();
      Serial.printf("[Heap] Free:%i, Size:%i, MinFree:%i, MaxAlloc:%i\n", ESP.getFreeHeap(), ESP.getHeapSize(), ESP.getMinFreeHeap(), ESP.getMaxAllocHeap());
  }

  

}

siegmar, 2021-07-21:
To use NimBle is the best way to solve your problem
cheers
Siegmar

VojtechBartoska, 2021-11-15:
@savagerex Hi, can I considered this as solved?

VojtechBartoska, 2022-03-30:
Closing as expired, version 1.0.4 is very old. We are not going to support that anymore. Thanks for your contribution.

closed 2022-03-30 by VojtechBartoska