espressif/arduino-esp32 #5435: error when build arduino-esp32 as a esp-idf component

tientu0001, 2021-07-21:
i used arduino-esp32 with esp-idf, for use library u8g2 with lcd ssd1306, follow
https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html
but when idf.py build. have problem, i cheked, var "esp_eth_phy_new_ksz8081" link with path lib "C:\esp\esp-idf\components\esp_eth\include\esp_eth.h" so don't have var esp_eth_phy_new_ksz8081. This var should link with "path/project/components\arduino\tools\sdk\esp32\include\esp_eth\include\esp_eth.h", right?
i use esp-idf 4.3, can someone help you? sorry about my english 

../components/arduino/libraries/WiFi/src/ETH.cpp:291:23: error: 'esp_eth_phy_new_ksz8081' was not declared in this scope
             eth_phy = esp_eth_phy_new_ksz8081(&phy_config);
                       ^~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/libraries/WiFi/src/ETH.cpp:291:23: note: suggested alternative: 'esp_eth_phy_new_ksz8041'
             eth_phy = esp_eth_phy_new_ksz8081(&phy_config);
[image]

tientu0001, 2021-07-21:
i try copy ... \components\arduino\tools\sdk\esp32\sdkconfig to main project but still same error

lbernstone, 2021-07-21:
Use esp-idf master branch.

florian-fouquet, 2021-07-21:
Hi,
I have exactly the same error using PlatformIO. I updated to espressif32 platform version 3.3.0 and i'm using the latest tag 2.0.0-rc1 of arduino-esp32.
I have replaced esp_eth_phy_new_ksz8081 by the suggested alternative esp_eth_phy_new_ksz8041 and compile with success.

But i don't know if ethernet is working well as i my project don't use it.

Could be great to find a fix in order to work with latest platformio espressif platform.

chegewara, 2021-07-21:
PlatformIO and arduino as component are 2 different things.
> Could be great to find a fix in order to work with latest platformio espressif platform.

Ask in platformio repository.

tientu0001, 2021-07-22:
> Use esp-idf master branch.

i install esp-idf master branch but new error,
" components/esp_eth/include/esp_eth_mac.h:400:9: error: expected primary-expression before '.' token .clock_config.rmii.clock_mode = EMAC_CLK_DEFAULT,
/components/arduino/libraries/WiFi/src/ETH.cpp:253:39: note: in expansion of macro 'ETH_MAC_DEFAULT_CONFIG"
do you know problem?
tks
[image]
[image]

me-no-dev, 2021-07-22:
There is currently a regression with ETHERNET and IDF Master. It should be fixed very soon. Try using https://github.com/espressif/esp-idf/commit/d93887f9f0388b90d1ba5940987a2d2e413b961f

tientu0001, 2021-07-22:
someone know what problem, help me....
i try both release verion, release branch and development branch, still problem when idf.py build

digibake, 2021-07-24:
@me-no-dev's response solved the problem for me. Thanks!

For anyone else hitting this issue,

1. I opened the esp-idf repository in github desktop ---- $HOME/esp/esp-idf (although you could just as easily use the command line if you like)
2. Found the commit d93887f ---- 8th July 2021 called "Merge branch 'feature/openthread-lib' into 'master'"
3. Created a new branch from this commit (see image below)
4. Closed the terminal and reopened it / reloaded idf.py
5. idf.py clean
6. idf.py build... and success.

[image]

VojtechBartoska, 2022-03-30:
Solved already, thanks for your contribution.

closed 2022-03-30 by VojtechBartoska