espressif/arduino-esp32 #5436: Cannot compile with 2.0.0RC1 in PlatformIO

garageeks, 2021-07-21:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Compiling an Arduino sketch in PlatformIO using the latest 2.0.0 RC1 release
Describe your system( Hardware, computer, O/S, core version, environment).
Windows 10 x64 version 20H2, Visual Studio Code 1.58.2, PlatformIO Core 5.1.1

I wanted to get latest version that has a patch for issue #3659, therefore I modified my platformio.ini project file with these environment variables:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
platform_packages = 
	toolchain-xtensa32 @ ~2.80400.0
	framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.0-rc1
```

Then compiling fails. I cleaned up, as well as closed and restarted Visual Studio Code. It looks like wrong compiler.
I tried to remove the toolchain-xtensa32 line in platformio.ini, and different errors appear.
I have platform Espressif32 2.1.0 installed, shouldn't it pick up the latest SDK and compiler?

Thank you in advance for your support
Nick

This is the compiler output:

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.4ada3f5
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Converting PSC-Gen2-FirmwareESP32.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit Unified Sensor> 1.1.4    
|-- <Adafruit BME280 Library> 2.1.2    
|   |-- <Adafruit Unified Sensor> 1.1.4
|   |-- <SPI> 2.0.0
|   |-- <Wire> 2.0.0
|-- <Adafruit ADXL343> 1.3.0
|   |-- <Adafruit BusIO> 1.7.1
|   |   |-- <Wire> 2.0.0
|   |   |-- <SPI> 2.0.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|   |-- <Adafruit Unified Sensor> 1.1.4
|-- <PMS Library> 1.1.0
|-- <Time> 1.6.0
|-- <Adafruit MCP23008 library> 2.1.0
|   |-- <Adafruit BusIO> 1.7.1
|   |   |-- <Wire> 2.0.0
|   |   |-- <SPI> 2.0.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <ArduinoJson> 6.17.2
|-- <Adafruit SI1145 Library> 1.1.1
|   |-- <Wire> 2.0.0
|-- <AsyncTCP-esphome> 1.1.1
|-- <RunningMedian> 0.3.3
|-- <AsyncElegantOTA> 2.2.5
|   |-- <AsyncTCP-esphome> 1.1.1
[... 106 lines cut ...]
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-misc.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-psram.c.o
Archiving .pio\build\esp32dev\libff2\libHTTPClient.a
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32dev\libFrameworkArduino.a
Linking .pio\build\esp32dev\firmware.elf
c:/users/proprietario/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: final link failed: bad value
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
===================================================== [FAILED] Took 36.53 seconds =====================================================The terminal process "C:\Users\proprietario\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32dev'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

garageeks, 2021-07-21:
I tried to upgrade espressif32 to latest version 3.3 but still no joy. It looks like this one uses ESP-IDF 4.3...  xtensa32-toolchain 8.4.0 appears to be latest version. Maybe it cannot compile until ESP-IDF 4.4 is installed? But I have no idea how to install it on PlatformIO

Here latest output

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.4ada3f5
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Converting PSC-Gen2-FirmwareESP32.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit Unified Sensor> 1.1.4
|-- <Adafruit BME280 Library> 2.1.2
|   |-- <Adafruit Unified Sensor> 1.1.4
|   |-- <SPI> 2.0.0
|   |-- <Wire> 2.0.0
|-- <Adafruit ADXL343> 1.3.0
|   |-- <Adafruit BusIO> 1.7.1
|   |   |-- <Wire> 2.0.0
|   |   |-- <SPI> 2.0.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|   |-- <Adafruit Unified Sensor> 1.1.4
|-- <PMS Library> 1.1.0
|-- <Time> 1.6.0
|-- <Adafruit MCP23008 library> 2.1.0
|   |-- <Adafruit BusIO> 1.7.1
|   |   |-- <Wire> 2.0.0
|   |   |-- <SPI> 2.0.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <ArduinoJson> 6.17.2
|-- <Adafruit SI1145 Library> 1.1.1
|   |-- <Wire> 2.0.0
|-- <AsyncTCP-esphome> 1.1.1
|-- <RunningMedian> 0.3.3
|-- <AsyncElegantOTA> 2.2.5
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <AsyncTCP-esphome> 1.1.1
|   |   |-- <FS> 2.0.0
|   |   |-- <WiFi> 2.0.0
|   |   |-- <ArduinoJson> 6.17.2
|   |-- <FS> 2.0.0
|   |-- <Update> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ArduinoJson> 6.17.2
|-- <HTTPClient> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <WiFiClientSecure> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <WiFi> 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\PSC-Gen2-FirmwareESP32.ino.cpp.o
Compiling .pio\build\esp32dev\src\src\EspSoftwareSerial\src\SoftwareSerial.cpp.o
Compiling .pio\build\esp32dev\src\src\ModbusMaster\src\ModbusMaster.cpp.o
Generating partitions .pio\build\esp32dev\partitions.bin
[... 80 lines cut ...]
Archiving .pio\build\esp32dev\libff2\libHTTPClient.a
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-matrix.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-misc.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32dev\libFrameworkArduino.a
Linking .pio\build\esp32dev\firmware.elf
c:/users/proprietario/.platformio/packages/toolchain-xtensa32@2.80400.210211/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: final link failed: bad value
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1

chegewara, 2021-07-22:
I can confirm it. With `arduino-esp32#2.0.0-alpha1` i can build, even for S2, but `rc1` fail to build.

HDPlayser, 2021-07-22:
Same Issue here.  Can't even link.

me-no-dev, 2021-07-22:
Yes, this is a "known" issue on which the PlatformIO guys are working. This is not an issue with this repository, but an issue with PIO.

Try adding `platform_packages = espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r1` to your project config.
If you need support for S2 and C3, packages are `espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1` and `espressif/toolchain-riscv32-esp @ 8.4.0+2021r1`

closed 2021-07-22 by me-no-dev

garageeks, 2021-07-22:
Hi @me-no-dev thank you for your feedback.
I have tried this `platform_packages` configuration:
```
platform_packages = 
	toolchain-xtensa-esp32 @ 8.4.0+2021r1
	toolchain-xtensa32 @ ~2.80400.0
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0-rc1
```
It seems to pickup a newer version of toolchain-xtensa32 but not toolchain-xtensa-esp32.
Nonetheless it still fails the same way
```
Linking .pio\build\esp32dev\firmware.elf
c:/users/proprietario/.platformio/packages/toolchain-xtensa32@2.80400.210211/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: final link failed: bad value
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

If I use this configuration:
```
platform_packages = 
	toolchain-xtensa-esp32 @ 8.4.0+2021r1
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0-rc1
```

Thank it ignores the toolchain and uses the default Espressif 3.3.0 toolchain (latest platform version available on PIO), that is missing some files:
```
Linking .pio\build\esp32dev\firmware.elf
c:/users/proprietario/.platformio/packages/toolchain-xtensa32@2.50200.97/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find crt1-sim.o: No such file or directory
c:/users/proprietario/.platformio/packages/toolchain-xtensa32@2.50200.97/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find _vectors.o: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

How can I force it to use `toolchain-xtensa-esp32 @ 8.4.0+2021r1` ?
In any case I'll open an issue on PIO GitHub so we can track their side.

Thank you
Nick

me-no-dev, 2021-07-22:
@valeros @ivankravets please take a look!

valeros, 2021-07-23:
Hi guys, as no one has submitted an issue to the PlatformIO repository I'll briefly answer here. In a nutshell, as we're approaching the core v2.0, PlatformIO's `espressif32` dev-platform is being gradually switched to the official packages provided by Espressif as well. This includes new package names, new versions of these packages, etc. During this transitional period, anyone who wants to use the upstream Arduino core from master, any release candidates, etc. need to install the dev-platform from a special branch that contains required changes, something like this:

```ini
[env:esp32-s2-saola-1]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
framework = arduino
board = esp32-s2-saola-1
```
This workflow is temporary and once the core v2.0 is officially released, everything will get back to normal. In case you still have any problem, please report them directly to https://github.com/platformio/platform-espressif32

garageeks, 2021-07-23:
Hi @valeros 
Thank you for answering, indeed the trick worked.
I changed my project platformio.ini as follows:
`[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
platform_packages = 
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0-rc1`

... and indeed now it compiles. Now, let's put 2.0RC1 up to its paces!
`Linking .pio\build\esp32dev\firmware.elf
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.6% (used 37960 bytes from 327680 bytes)
Flash: [========= ]  87.6% (used 1148149 bytes from 1310720 bytes)
Building .pio\build\esp32dev\firmware.bin
esptool.py v3.1
Merged 25 ELF sections`

rinscr3003, 2021-07-24:
This also cannot be compiled in Arduino IDE. With the same error.
@me-no-dev

rinscr3003, 2021-07-25:
Solved.
I changed the toolchain version to 2021r1.
@me-no-dev

ullix, 2021-08-11:
@valeros I tried your ini settings from https://github.com/espressif/arduino-esp32/issues/5436#issuecomment-885469600 but got this response:
```
Tool Manager: Installing platform-espressif32 @ *
Error: Could not find the package with 'platform-espressif32 @ *' requirements for your system 'linux_x86_64'
```
Is this valid only for Windows? I there a Linux version also available?

ivankravets, 2021-08-11:
@ullix do you use the latest PlatformIO Core version? `pio ––version`

Please check also Internet connection

ullix, 2021-08-11:
```
$ pio --version
PlatformIO Core, version 5.1.1
```
Just reinstalled after this problem: https://community.platformio.org/t/pio-upgrading-actually-downgraded/22876

ullix, 2021-08-11:
Ooops, sorry, my mistake: I put the line under `platform_packages` not under `platform`! 

At the correct position it compiles, but linking fails with:
```
/home/ullix/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/lib741/libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x8): undefined reference to `mbedtls_md5_starts'
```
Is a different `ESP Async WebServer` needed? If so, how can I get it?