espressif/arduino-esp32 #5437: Need help with 'camera_config_t has no member named grab_mode'

russellsim, 2021-07-22:
### Hardware:
Board:						       AI Thinker ESP32-CAM
Core Installation version:		       Same issue on both 2.0.0-rc1 and 1.06
IDE name:					       Arduino IDE
Flash Frequency:				       80Mhz
PSRAM enabled:                                     yes
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
This is really strange because it's defined at https://github.com/espressif/arduino-esp32/blob/4f9e583b29040bcefabd54d981c38324abb2ee7c/tools/sdk/esp32/include/esp32-camera/driver/include/esp_camera.h

Any idea what is wrong?

### Sketch: 
```cpp

#include <esp_camera.h>

void setup() {
  camera_config_t config;
  config.grab_mode = CAMERA_GRAB_LATEST;
}

void loop() {
}
```

### Error Message:
```
C:\Users\Me\Documents\Arduino\camera_issue\camera_issue.ino: In function 'void setup()':
camera_issue:5:10: error: 'struct camera_config_t' has no member named 'grab_mode'
   config.grab_mode = CAMERA_GRAB_LATEST;
          ^
camera_issue:5:22: error: 'CAMERA_GRAB_LATEST' was not declared in this scope
   config.grab_mode = CAMERA_GRAB_LATEST;
                      ^
exit status 1
'struct camera_config_t' has no member named 'grab_mode'
```

me-no-dev, 2021-07-22:
is it possible that you have two versions of the framework installed? Maybe you have the board manager version and also the github version in `C:\Users\Me\Documents\Arduino\hardware`?

russellsim, 2021-07-22:
Yes, that has helped me to solve the issue. 
Cleaning everything up and ensuring that I've only got 2.0.0-rc1 has solved the issue.

closed 2021-07-22 by russellsim

karff, 2023-03-07:
> 是的，这帮助我解决了这个问题。清理所有内容并确保我只有 2.0.0-rc1 已经解决了这个问题。
Excuse me, how did you find this redundant framework?