espressif/arduino-esp32 #5440: ledcWriteTone affects both channels

spyder0069, 2021-07-22:
I am using the two pwm channels on a ESP32 to control a piezo speaker on channel 0 and the backlight on a LCD on channel 1.  What I have come across is that after using the ledcWriteTone my backlight dims.  I think that command is adjusting the frequency for both channels.  I am not real familiar with how that hardware works.  If I call my ledcSetup for channel 1 I get my brightness back but I get a blink in my brightness with this.

So my questions:
Is ledcWriteTone to one channel supposed to affect the other?

If so any ideas of a creative work around so that I can use pwm on my backlight channel without be affected by the speaker?

Appreciate any advice!

lbernstone, 2021-07-22:
The [ESP32Servo](https://www.arduino.cc/reference/en/libraries/esp32servo/) library is much better about mixing and matching pwm functions.

spyder0069, 2021-07-22:
> The [ESP32Servo](https://www.arduino.cc/reference/en/libraries/esp32servo/) library is much better about mixing and matching pwm functions.

Thank you for the reply.  Initially I tried that library for just the tone function (still using ledC for the backlight) and still had the problem but I just tried it using both tone and pwm from the library and they are playing nicely together.  I appreciate the help!

SuGlider, 2021-07-23:
Closing the issue. Please fell free to reopen it if necessary.

closed 2021-07-23 by SuGlider

spyder0069, 2022-04-05:
Just coming back to address a fix for anyone having this issue.  The esp32servo library did originally solve the problem but I tried to tackle this again and found the solution.  Apparently the esp32 pwm channels are grouped together in two groups.  I assigned my buzzer to channel 1 and then my lcd backlight to channel 9 and now everything is happy using the built in ledcSetup and ledcWriteTone commands.  Don't quote me on this but I think each group shared the frequency so when the tone command is changing the frequency for sound it was also affecting the frequency on my other pin being used for backlight.  By putting them in seperate groups it isolates the frequency for each group.  That is my guess anyway.  :^)

fillybrese, 2022-04-05:
> Just coming back to address a fix for anyone having this issue. The esp32servo library did originally solve the problem but I tried to tackle this again and found the solution. Apparently the esp32 [...]

@spyder0069 thanks for the hint about the channel groups. also i noticed the different ledc channels affecting each others' frequencies when using ledcWriteTone. channel 0 affected channel 1 in my case. i moved channel 1 to channel 9 and it does not have the effect anymore, but unfortunately i can't get it working at all on channel 9.

Anyways seems there's certainly this affectation for library users to be aware of, thanks for opening the ticket!

spyder0069, 2022-04-05:
Here is some sample of my working code:
`
#define TFT_BACKLIGHT   32
byte buzzer=2;  //set to pin 2

void setup{
 pinMode(TFT_BACKLIGHT, OUTPUT);
 digitalWrite(TFT_BACKLIGHT, LOW);
 setlcdbrightness(70);

  ledcDetachPin(buzzer); //needed to unhook the buzzer from FW33 and below otherwise will give solid tone until the first beep 
routine is done.
  pinMode(buzzer, OUTPUT);
  digitalWrite(buzzer, LOW);
}

void beep(word tonefrequency, word toneduration, word tonepause){
  ledcSetup(1,2000,8);
  ledcWrite(1, 255); //ledcWrite(uint8_t channel, uint32_t duty);
  ledcAttachPin(buzzer,1);  //ledcAttachPin(uint8_t pin, uint8_t channel);
  ledcWriteTone (1, tonefrequency); //ledcWriteTone(uint8_t channel, double freq);
  delay(toneduration);
  ledcWriteTone (1, 0);

  tonepause=tonepause/2;  // i don't remember why I was halving the pause here.
  delay(tonepause);   //this pause in between notes
}

void setlcdbrightness(byte lcddutycycle){
 //lcddutycycle equals 0-255
  ledcSetup(9,2000,8); //ledcSetup(PWM1_Ch, PWM1_Freq, PWM1_Res);  USE CHANNEL 9 because pwm in esp is split into 16 channels and 2 groups.  This way buzzer and lcd are in different groups and don't conflict.
  ledcAttachPin(TFT_BACKLIGHT,9); //ledcAttachPin(LED_GPIO, PWM1_Ch);
  ledcWrite(9, lcddutycycle); //ledcWrite(PWM1_Ch, PWM1_DutyCycle++);   
}

`