espressif/arduino-esp32 #5444: cannot read SSID from SD file correctly - hardcoded SSID workes fine

dsyleixa, 2021-07-23:
I cannot read an SSID String from an SD file correctly - hardcoded it workes fine though:

SD file content:
```
(...)
SSID
WLAN_ABCDE
(...)
```
code to read from SD:
```
String SSID     = "WLAN-ABCD";  //default
pwd = "vwxyz";

void setup() {
         //...
         String buf1="", bufold="";
         Serial.println("reading SD config.txt...");
         while (SD_File.available()) {
            buf1 = SD_File.readStringUntil('\n');            
            Serial.print("SD-buf1 value=");
            Serial.println(buf1); //Printing for debugging purpose            
            if(bufold.startsWith((String)"SSID" ) ) { 
               //SSID=buf1;  // <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< optionally un/out-comment!
               Serial.print("SD-SSID value=");
               Serial.println(SSID);  //Printing for debugging purpose
            }
            bufold=buf1;
         }
         SD_File.close();
         //...
```
the SD read code functions well, 
Serial.print-out:
```
SD-buf1 value=SSID;
SD-buf1 value=WLAN-ABCDE;
SD-SSID value=WLAN-ABCDE
```
(which is correct and apparently identical to the default string)

When I start WiFi with the default (outcommented // SSID=buf1; ), WiFi starts fine:
```
   display.print("try WiFi");
   WiFi.mode(WIFI_STA);
   WiFi.config(LOCALIP, GATEWAY, SUBNET, GATEWAY, GATEWAY);
   WiFi.begin( SSID.c_str(), pwd );
```
WiFi becomes connected and then starts WiFiServer and WebServer, too.

But when I un-comment the line and assign the String from SD by SSID=buf1, then WiFi does not start and just prints millions of "....."

It seems as if the String read from SD contains (invisible?) extra bytes?
What am I missing?
How to fix it?

felmue, 2021-07-23:
Hello @dsyleixa 

your default SSID in your code is `WLAN-ABCD` whereas the SSID read from SD card is `WLAN-ABCDE`. It has an additional 'E' at the end. Could that be your issue?

Thanks
Felix

dsyleixa, 2021-07-23:
ah, no, that is just a c+p mistake, but thanks for your attentiveness!

dsyleixa, 2021-07-23:
ah, no, that is just a c+p mistake, but thanks for your attentiveness!
(the "real" SSID in my original code is quite different actually, but identical in either location )

dsyleixa, 2021-07-23:
solved:
the SD was written with Windows Editor which adds \r\n as line delimiters
So I had to 1st read until '\r', to get the line, and then an extra read until '\n' to drop the newline char:
```
while (SD_File.available()) {
            buf1 = SD_File.readStringUntil('\r');
            SD_File.readStringUntil('\n');
            Serial.println();
            Serial.println("SD-buf1 value=" + buf1);
            if(bufold.startsWith((String)"SSID" ) ) { //myString.startsWith(myString2)
               SSID=buf1;
               Serial.println("SD-SSID value="+SSID+(String)"<");
            }
            //...

```
Now I got it working!

regrettable though that (String).readStringUntil() is not designed to work smarter ...
and JTM: also regrettable that String don't provide more methods for substrings (similar like string.h does) ...

lbernstone, 2021-07-23:
std::string is available if you prefer.  arduino-esp32 has String to match up with the Arduino-api, but esp-idf is always under the covers, and has a pretty complete stdlib implementation.

dsyleixa, 2021-07-24:
yes, thanks, but for std::string also sth like .readln() or .readStringUntil(EOL) are missing (please CMIIW),
additional I also don't find  .hasSubstr() or .insert(substr, pos, len) or .delete(pos, len) or .msgToken(token, delimiters)
But that's perhaps more another topic.

dsyleixa, 2021-07-24:
yes, thanks, but for std::string also sth like .readln() or .readStringUntil(EOL) are missing (please CMIIW),
additional I also don't find .posSubstr() or .insert(substr, pos, len) or .delete(pos, len) or .msgToken(token, delimiter1, delimiter2)
But that's probably more another topic.

Perhaps someone has a better advice how to read an entire textline all at once until end of line, discarding any EoL/CR?

closed 2021-07-25 by dsyleixa