espressif/arduino-esp32 #5445: Getting [e][wifigeneric.cpp:739] hostbyname(): dns failed when performing POST request

pzpsofficial, 2021-07-23:
Hi!

I am getting started with electronics and microcontrollers programming. 

I have made a simple circuit where I use DHT22 sensor to measure temperature and humidity. I have also made my own API in Node and Express (MongoDB as a database). It is very simple API, just two endpoints: one for getting. and one for posting data. I ma able to create succesfull requests using Postman (and with the browser). 

Just to be clear, I want to send temperature and humidity to my API, and then do some work with this data on Vue website (I think that this is not relevant at all, but again, just to be clear what I am trying to achieve). 

Now I will say what I am using:

- Windows 10 OS
- NodeMCU ESP32 microcontroller
- DHT22 sensor
- HTTPClient library (I think this one is causing a problem)
- PlatformIO with Arduino.h header file

Everything is ok, but when I try to send data to my database things fail. I am ALWAYS getting following error 

```[e][wifigeneric.cpp:739] hostbyname(): dns failed```

I have tried to make the POST request using both ```http://localhost:3000/endpoint``` and ```http://127.0.0.1/endpoint``` (that part is really strange, why I am getting DNS error when using IP address without domain name?). 

I have already looked up for the solution on the web. I have come across many similar questions on github, but any of them worked for me). Also, none of them were solving error caused by line ```739```. 

Here I will leave my code. Is is simple and short, so I will post all of it. Please, do not blame me for my c++ skills, I am getting better :D 

Thank you in advance, have a nice day or evening. 

Kind regards, Bartek.

```
#include <Arduino.h>
#include <DHT.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <ArduinoJson.h>
#include <HTTPClient.h>
#include <ArduinoJson.h>

#define DHTPIN 22
#define DHTTYPE DHT22

DHT dht(DHTPIN, DHTTYPE);

const char ssid[] = "MY_SSIID";
const char password[] = "MY_WIFI_PASSWORD";
const char apiAddress[] = "http://localhost:3000/reading";

unsigned long lastTime = 0;
unsigned long timerDelay = 60000;

struct sensorReading {
  float temperature;
  float humidity;
  float heatIndex;
};

sensorReading readDataFromSensor();
void sendData(sensorReading * reading);

void setup() {
  Serial.begin(9600);
  dht.begin();

  WiFi.begin(ssid, password);

 WiFi.config(IPAddress(192, 168, 1, 223), IPAddress(192, 168, 1, 1), IPAddress(255, 255, 255, 0));

  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print('.');
  }

}

void loop() {
  if (millis() > lastTime + timerDelay) {
    sensorReading data;
    sensorReading * p_data = NULL;
    data = readDataFromSensor();
    p_data = &data;
    sendData(p_data);
  }
}

sensorReading readDataFromSensor() {
  float temperature = dht.readTemperature();
  float humidity = dht.readHumidity();
  float heatIndex = dht.computeHeatIndex(temperature, humidity, false);

  if (isnan(temperature) || isnan(humidity)) {
    return sensorReading {};
  }

  const sensorReading dataToReturn {temperature, humidity, heatIndex};
  return dataToReturn;
}

void sendData(sensorReading * reading) {
  using namespace std;

  if(WiFi.status() == WL_CONNECTED) {
    HTTPClient http;

    http.begin(apiAddress, 3000);

    http.addHeader("Content-Type", "application/json");

    DynamicJsonDocument doc(1024);

    doc["temperature"] = reading->temperature;
    doc["humidity"] = reading->humidity;
    doc["heatIndex"] = reading->heatIndex;

    String dataToSend; 

    serializeJson(doc, dataToSend);

    int responseCode = http.POST(dataToSend);

    Serial.println(responseCode);
    http.end();
  } else {
    Serial.print("Ups...");
  }

  lastTime = millis();
}

```

felmue, 2021-07-24:
Hello @pzpsofficial 

`http.begin()` is overloaded multiple times; however non of these overloaded functions seem to take `url` and `port` as parameter. And since your `apiAddress` already contains the port number, try it without the port, e.g.
```
  http.begin(apiAddress);
```
Also check out this tutorial I found [here](https://randomnerdtutorials.com/esp32-http-get-post-arduino/).

Thanks
Felix

pzpsofficial, 2021-07-24:
Hello @felmue 

I have already tried: 

```
http.begin(apiAddress);
``` 

But I was getting other error, and github issue for that error was suggesting to pass port number as a second argument to http.begin function.

I was using tutorial provided by you as well 😅

Thanks for your time and willing to help!

Thanks,
Bartek

pzpsofficial, 2021-07-24:
@femule 

You were right I think. I removed port number from arguments list, and now I am getting another error...

```
[E][WiFiClient.cpp:258] connect(): socket error on fd 54, errno: 104, "Connection reset by peer"
```

I think that it is worth noting, that now I get http response code: ```-1``` in this case. 

Thanks, 
Bartek

garageeks, 2021-07-26:
I don't think "http://localhost:3000/reading" is a valid address. You should point to the IP address of the computer the database is running on, I'd say an IP address of the 192.168.1.x subnet
Of course Postman will work as it is running on the same computer which is hosting the database

pzpsofficial, 2021-07-27:
@garageeks 

That has solved my issue, thanks for help!

closed 2021-07-27 by pzpsofficial