espressif/arduino-esp32 #5446: MDNS fails the majority of the time (HelloServer included)

witnessmenow, 2021-07-24:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.6
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
MDNS does not work reliably for me on an ESP32 at all.

With the helloServer example running on my ESP32, waiting for it to be ready in the serial monitor I opened 30 tabs to "http://esp32.local" one by one with a few seconds in between and the first 7 failed, next 13 worked but the last 10 failed. so 17 out of 30 failed.

[image]

I ran the same test on helloServer example from ESP8266 with 20 tabs and they all loaded perfectly

[image]

I did not modify the examples other than putting my wifi credentials in.

The screen shots are from firefox, but I have the same results in chrome.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
[Hello Server example reproduces the error for me](https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples/HelloServer)

### Debug Messages:
There is no error message, nothing shows up in the logs at all, even with Debug

```
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START

....[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.91, MASK: 255.255.255.0, GW: 192.168.1.1
.
Connected to Mikrotik
IP address: 192.168.1.91
MDNS responder started
HTTP server started
///////// Made multiple requests using http://esp32.local here, below was using IP address
[E][WebServer.cpp:633] _handleRequest(): request handler not found
```

Is there anything you can suggest to try out? I could understand if had the same issues with ESP8266 boards that it has to be something in my network or on my PC that is the issue, but since it works find for the ESP8266 (i've also had no issues with other getting to raspi etc), it seems like an ESP32 issue

me-no-dev, 2021-07-24:
This does not look like mDNS issue. You computer should have cached the MDNS address and should not be asking the ESP anymore. Further more, you are getting an error in the web server itself

witnessmenow, 2021-07-24:
I'll do some more testing later with more devices (my Mac) but I only have mdns issues with esp32, I'll admit I know basically nothing about how it works but there has to be something unique to how the esp32 handle it if they are the only things that don't work correctly

VojtechBartoska, 2022-03-30:
Hello @witnessmenow, sorry for replying that late but did you solve your issue?

closed 2022-04-13 by VojtechBartoska