espressif/arduino-esp32 #5447: Connecting to 2 or multiple BT serial devices

emard, 2021-07-24:
I'd like to connect and communicate at the same
time with 2 different bluetooth devices
each device is providing a single serial port.

Is there some code example how to do it
with BluetoothSerial?

emard, 2021-10-02:
I have seen some activity in the source
maybe someone else has tried to connect 2 devices,
I'd like code example

digamesystems, 2021-12-22:
Hi. Anyone have any thoughts on this issue? 

Connecting to one device works great, but I'd also like to make multiple Bluetooth classic connections to devices so I can aggregate some sensor readings. 

If I simply copy the example code for a single connection and create multiple bluetooth connections, I get an indication that connecting to each of them has succeeded, but when data is received, it is only coming from the last bluetooth device I connected to.  

I'm running my app within the Arduino IDE using the .available() function.

```
 if (btUART2.available()) {
    processMessage(btUART2, 1); 
  }

  if (btUART1.available()) {
    processMessage(btUART1, 0); 
  } 
```

Here I only see .available() returning true for btUART2 -- the last bluetooth UART I connected to. If I reverse the order of the .connect call, I only see data from btUART1. 

Any help would be appreciated. Thanks!

khaled147, 2022-01-20:
I haven't tested connecting multiple BLE devices simultaneously but I was able to pull data from different sensors by cycling between disconnecting and connecting to the address of the device I want to connect to. Maybe consider that if it works.

digamesystems, 2022-01-20:
> I haven't tested connecting multiple BLE devices simultaneously but I was able to pull data from different sensors by cycling between disconnecting and connecting to the address of the device I want to connect to. Maybe consider that if it works.

Yeah, I can get that to work as well but I'd like it if the sensors could just send me data asynchronously without my having to poll them. Having the `.available()` function is really nice for situations where a sensor wakes up and transmits only when it has something to say...

VojtechBartoska, 2022-04-07:
Hello, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? 

Take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

digamesystems, 2022-04-07:
> Hello, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? ...

Thanks, @VojtechBartoska . I'll give it a try once my current release is out. 

I read through the release notes for 2.0.3-RC1 and didn't see anything directly related to the Bluetooth Serial library except for this issue: https://github.com/espressif/arduino-esp32/pull/6109

This only seems to be getting rid of a compiler warning...

Am I missing something? -- Thanks again for the suggestion!

VojtechBartoska, 2022-04-08:
There have been some changes in ESP-IDF itself which can have an influence on this.

digamesystems, 2022-04-08:
Thanks, @VojtechBartoska. 

I'll give it a try as soon as I get my release out the door. 

I’ll also dig into the source for the BluetoothSerial library and see how it’s structured. I suspect there might be a single, global pointer that’s being used to reference the port. This would explain the behavior I'm seeing where the last port to be declared is the one that gets the callbacks. -- 

Thinking about it, it might just be an architectural issue, where sharing of the Bluetooth hardware resource isn't supported.  After all, I don't think you can create multiple, independent instances of Serial with their own callbacks.

VojtechBartoska, 2022-05-05:
2.0.3 stable is out, any chance you gave it a shot?

VojtechBartoska, 2022-07-26:
@digamesystems Any news?

VojtechBartoska, 2022-08-23:
Closing this issue as expired due to no answer, if needed you can reopen it.

closed 2022-08-23 by VojtechBartoska