espressif/arduino-esp32 #5448: Can't compile when using I2C

felix900006, 2021-07-24:
### Hardware:
Board:							ESP32 Dev Module
Espressif 32 Core Installation version:			3.3.0
IDE name:						Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I get an copile error every time i try to use I2C

### Sketch: 
```cpp

#include <Arduino.h>
#include <Ticker.h>
#include "SparkFunHTU21D.h"
#include <SPI.h>
#include <Wire.h>
#include <QuickPID.h>
#include <StopWatch.h>
#include <WiFi.h>
#include <ESPmDNS.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncElegantOTA.h>
#include <FS.h>
#include <SPIFFS.h>
#include <ArduinoJson.h>

int outputPin = 21;

AsyncWebServer server(80);
AsyncWebSocket ws("/ws");

const char *ssid = "****";
const char *key = "****";
const char *hostname = "ESP32";
const char *mdns_name = "ESP32";

HTU21D htu;

const int16_t sampleTimeUs = 10000;
const int outputMax = 255;
const int outputMin = 180;

bool printOrPlotter = 0;
float POn = 1.0;
float DOn = 0.0;

byte outputStep = 5;
byte hysteresis = 1;

int setpoint = 80;
int output = 0;

float Input, Output, Setpoint;
float Kp = 0, Ki = 0, Kd = 0;
bool pidLoop = false;
static bool computeNow = false;
void runPid();

Ticker timer1(runPid, sampleTimeUs, 0, MICROS_MICROS);
QuickPID PID = QuickPID(&Input, &Output, &Setpoint, Kp, Ki, Kd, POn, DOn, QuickPID::DIRECT);

// Measurement averaging
float avg(float inputVal) {
  static float arrDat[16];
  static int pos;
[... 180 lines cut ...]
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-touch.c.o
C:\Users\Deatsch\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c: In function 'spiTransferBytesNL':
C:\Users\Deatsch\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
C:\Users\Deatsch\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32doit-devkit-v1\libFrameworkArduino.a
Indexing .pio\build\esp32doit-devkit-v1\libFrameworkArduino.a
Linking .pio\build\esp32doit-devkit-v1\firmware.elf
.pio\build\esp32doit-devkit-v1\libFrameworkArduino.a(esp32-hal-i2c.c.o):(.iram1.24.literal+0x0): undefined reference to `xEventGroupSetBitsFromISR'
.pio\build\esp32doit-devkit-v1\libFrameworkArduino.a(esp32-hal-i2c.c.o): In function `i2cIsrExit':
C:\Users\Deatsch\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-i2c.c:838: undefined reference to `xEventGroupSetBitsFromISR'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32doit-devkit-v1\firmware.elf] Error 1
```

BaooQ, 2022-02-07:
Have you solved the problem? I think I meet the same question...

VojtechBartoska, 2022-02-07:
@BaooQ Are you on the latest 2.0.2 Arduino core ESP32 version?

BaooQ, 2022-02-07:
@VojtechBartoska This one? I'm not sure.
[image]

VojtechBartoska, 2022-02-07:
It's old one.

Latest is available here: https://github.com/espressif/arduino-esp32/releases/tag/2.0.2

Take a look on installing guide: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

BaooQ, 2022-02-07:
@VojtechBartoska Thank you so much!

BaooQ, 2022-02-09:
I think the problem was solved.
I check the code,find this in  **event_groups.h** :

#if( configUSE_TRACE_FACILITY == 1 )
	BaseType_t xEventGroupSetBitsFromISR( EventGroupHandle_t xEventGroup, const EventBits_t uxBitsToSet, BaseType_t *pxHigherPriorityTaskWoken );
#else
	#define xEventGroupSetBitsFromISR( xEventGroup, uxBitsToSet, pxHigherPriorityTaskWoken ) xTimerPendFunctionCallFromISR( vEventGroupSetBitsCallback, ( void * ) xEventGroup, ( uint32_t ) uxBitsToSet, pxHigherPriorityTaskWoken )
#endif

And I remembered  i set the **configUSE_TRACE_FACILITY**  as 1 in some time before, to use some functions in rtos.h ,I try to set it back to 0 and then it works .

VojtechBartoska, 2022-02-23:
@BaooQ Hi, glad you found a solution. I'm closing the issue. If needed, you can reopen it.

closed 2022-02-23 by VojtechBartoska