espressif/arduino-esp32 #5450: [Question]: initialisation from imcompatible pointer Type

hasenradball, 2021-07-25:
### Hardware:
Board:						ESP32 Dev Module
Core Installation version:			1.0.6
IDE name:				        Platform.io
Flash Frequency:			       40Mhz
PSRAM enabled:                              
Upload Speed:					no upload 
Computer OS:                                  Windows 10

### Description:
Question:

Is this issue known?

[image]

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

VojtechBartoska, 2022-03-29:
Hello @hasenradball, sorry for replying so late. Any chance you will test this on latest version 2.0.2? Thanks a lot in advance for your answer.

hasenradball, 2022-03-29:
Hi,
yes I could test this with the actual version.

hasenradball, 2022-03-29:
I testetd it on the latest version of the Espressif 32 on PIO.
`PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module`

The issue is still available...
see below.

```

> Executing task: C:\Users\Frank\.platformio\penv\Scripts\platformio.exe run --environment esp32dev <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-openocd-esp32 2.1000.20210721 (10.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <cVCC_local>
|-- <WiFi> 1.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\liba03\cVCC_local\cVCC_local.cpp.o
Compiling .pio\build\esp32dev\lib170\WiFi\ETH.cpp.o
Compiling .pio\build\esp32dev\lib170\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32dev\lib170\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32dev\lib170\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32dev\lib170\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32dev\lib170\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32dev\lib170\WiFi\WiFiSTA.cpp.o
Archiving .pio\build\esp32dev\liba03\libcVCC_local.a
Compiling .pio\build\esp32dev\lib170\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32dev\lib170\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32dev\lib170\WiFi\WiFiUdp.cpp.o
Indexing .pio\build\esp32dev\liba03\libcVCC_local.a
Archiving .pio\build\esp32dev\libFrameworkArduinoVariant.a
Compiling .pio\build\esp32dev\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o
Indexing .pio\build\esp32dev\libFrameworkArduinoVariant.a
Compiling .pio\build\esp32dev\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Print.cpp.o
Archiving .pio\build\esp32dev\lib170\libWiFi.a
Compiling .pio\build\esp32dev\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\WMath.cpp.o
Indexing .pio\build\esp32dev\lib170\libWiFi.a
Compiling .pio\build\esp32dev\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\base64.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-adc.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-bt.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-log.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-matrix.c.o
[... 10 lines cut ...]
Compiling .pio\build\esp32dev\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\main.cpp.o
C:\Users\Frank\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c: In function 'spiTransferBytesNL':
C:\Users\Frank\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
C:\Users\Frank\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling .pio\build\esp32dev\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32dev\libFrameworkArduino.a
Indexing .pio\build\esp32dev\libFrameworkArduino.a
Linking .pio\build\esp32dev\firmware.elf
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.6% (used 37876 bytes from 327680 bytes)
Flash: [=====     ]  49.7% (used 650914 bytes from 1310720 bytes)
Building .pio\build\esp32dev\firmware.bin
esptool.py v3.1
Merged 1 ELF section
================================================================= [SUCCESS] Took 45.41 seconds =================================================================
```

me-no-dev, 2022-03-29:
@hasenradball  PIO's original support is for very old version now. Check: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-platformio

hasenradball, 2022-03-30:
Hi @me-no-dev,

I changed now to the actual version, and get mor issues:

```
Processing esp32dev (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream; board: esp32dev; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)       
PACKAGES:
 - framework-arduinoespressif32 2.0.3+sha.1e388a2
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-openocd-esp32 2.1000.20210721 (10.0)
 - toolchain-xtensa-esp32 8.4.0+2021r1
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <cVCC_local>
|-- <WiFi> 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Compiling .pio\build\esp32dev\lib170\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\USBMSC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\base64.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-adc.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-bt.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-dac.c.o
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp: In function 'void _arduino_event_cb(void*, esp_event_base_t, int32_t, void*)':
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:289:41: warning: unused variable 'event' [-Wunused-variable]
      wifi_event_sta_authmode_change_t * event = (wifi_event_sta_authmode_change_t*)event_data;
                                         ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:294:35: warning: unused variable 'event' [-Wunused-variable]
      wifi_event_sta_connected_t * event = (wifi_event_sta_connected_t*)event_data;
                                   ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:299:38: warning: unused variable 'event' [-Wunused-variable]
      wifi_event_sta_disconnected_t * event = (wifi_event_sta_disconnected_t*)event_data;
                                      ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:304:28: warning: unused variable 'event' [-Wunused-variable]
         ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
                            ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:316:35: warning: unused variable 'event' [-Wunused-variable]
      wifi_event_sta_scan_done_t * event = (wifi_event_sta_scan_done_t*)event_data;
                                   ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:331:34: warning: unused variable 'event' [-Wunused-variable]
   wifi_event_ap_probe_req_rx_t * event = (wifi_event_ap_probe_req_rx_t*)event_data;
                                  ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:336:33: warning: unused variable 'event' [-Wunused-variable]
   wifi_event_ap_staconnected_t* event = (wifi_event_ap_staconnected_t*) event_data;
                                 ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:341:36: warning: unused variable 'event' [-Wunused-variable]
   wifi_event_ap_stadisconnected_t* event = (wifi_event_ap_stadisconnected_t*) event_data;
                                    ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:346:36: warning: unused variable 'event' [-Wunused-variable]
      ip_event_ap_staipassigned_t * event = (ip_event_ap_staipassigned_t*)event_data;
                                    ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:356:23: warning: unused variable 'eth_handle' [-Wunused-variable]
      esp_eth_handle_t eth_handle = *(esp_eth_handle_t *)event_data;
                       ^~~~~~~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:369:28: warning: unused variable 'event' [-Wunused-variable]
         ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
                            ^~~~~
[... 27 lines cut ...]
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp:99:5: error: missing initializer for member 'tone_msg_t::frequency' [-Werror=missing-field-initializers]
[... cut ...]
-------------  --------  ------------
esp32dev       FAILED    00:00:23.868
==================================================================== 1 failed, 0 succeeded in 00:00:23.868 ====================================================================Der Terminalprozess "C:\Users\Frank\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32dev'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

```

me-no-dev, 2022-03-30:
@hasenradball it is very much possible that since we added S3 support, support in PIO degraded. I will poke @Jason2866 here for ideas. esptool and the toolchain are not the correct version. PIO guys are from Ukraine and understandably they are in a sticky situation ATM, so we need to resort to "hacks" to keep things going.

cc: @pedrominatel we might need update of the docs

Jason2866, 2022-03-30:
@hasenradball  This setup should do the job. It uses Arduino core 2.0.3-rc1
```
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
```

hasenradball, 2022-03-31:
Thanks for the reply,
I will check this again in the evening.

hasenradball, 2022-03-31:
Got the following errorwith this configuration:

```

[env:esp32dev]
platform = platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32dev
monitor_speed = 115200
framework = arduino
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master

upload_protocol = esptool
upload_speed = 921600
upload_port = COM6
```

```
> Executing task: C:\Users\Frank\.platformio\penv\Scripts\platformio.exe run --environment esp32dev <

Processing esp32dev (platform: platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Error: Traceback (most recent call last):
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\platformio\__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 134, in cli
    process_env(
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 176, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 67, in process
    result = init_platform(self.options["platform"]).run(
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 322, in init_platform
    return PlatformFactory.new(name)
  File "C:\Users\Frank\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 70, in new
    platform_cls = getattr(
AttributeError: module 'platformio.platform.platform' has no attribute 'PlatformPlatform'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

Der Terminalprozess "C:\Users\Frank\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32dev'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

```

Jason2866, 2022-03-31:
The setup is wrong. Either you do
```
[env:esp32dev]
platform = platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32dev
monitor_speed = 115200
framework = arduino
upload_protocol = esptool
upload_speed = 921600
upload_port = COM6
```
or when you want to use latest git master
```
[env:esp32dev]
platform = platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32dev
monitor_speed = 115200
framework = arduino
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master

upload_protocol = esptool
upload_speed = 921600
upload_port = COM6
```
The `.git` is mandantory !!

hasenradball, 2022-03-31:
@Jason2866  sorry I can't chatch your measage!

What do you mean with `.git`:
You copied exactly the same as I:
My config looks like that now:

```
[env:esp32dev]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32dev
monitor_speed = 115200
framework = arduino
;platform_packages =
;   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master

upload_protocol = esptool
upload_speed = 921600
upload_port = COM6
```

Jason2866, 2022-03-31:
I tried this setup. I does work. Your Platformio setup is messed up. Delete the hidden folder `.platformio`

hasenradball, 2022-03-31:
OK I deleted the `.platformio`folder:

Thats my result after compile, many issues:

```
> Executing task: C:\Users\Frank\.platformio\penv\Scripts\platformio.exe run --environment esp32dev <

Processing esp32dev (platform: https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
FILESYSTEM  spiffs
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.3) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)        
PACKAGES:
 - framework-arduinoespressif32 2.0.3+sha.ab197e1
 - tool-esptoolpy 1.30300.220323 (3.3.0)
 - tool-mklittlefs 1.203.210628 (2.3)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-openocd-esp32 2.1000.20210721 (10.0)
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <cVCC_local>
|-- <WiFi> 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\lib170\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Stream.cpp.o
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp: In function 'void _arduino_event_cb(void*, esp_event_base_t, int32_t, void*)':
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:289:41: warning: unused variable 'event' [-Wunused-variable]
      wifi_event_sta_authmode_change_t * event = (wifi_event_sta_authmode_change_t*)event_data;
                                         ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:294:35: warning: unused variable 'event' [-Wunused-variable]
      wifi_event_sta_connected_t * event = (wifi_event_sta_connected_t*)event_data;
                                   ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:299:38: warning: unused variable 'event' [-Wunused-variable]
      wifi_event_sta_disconnected_t * event = (wifi_event_sta_disconnected_t*)event_data;
                                      ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:304:28: warning: unused variable 'event' [-Wunused-variable]
         ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
                            ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:316:35: warning: unused variable 'event' [-Wunused-variable]
      wifi_event_sta_scan_done_t * event = (wifi_event_sta_scan_done_t*)event_data;
                                   ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:331:34: warning: unused variable 'event' [-Wunused-variable]
   wifi_event_ap_probe_req_rx_t * event = (wifi_event_ap_probe_req_rx_t*)event_data;
                                  ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:336:33: warning: unused variable 'event' [-Wunused-variable]
   wifi_event_ap_staconnected_t* event = (wifi_event_ap_staconnected_t*) event_data;
                                 ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:341:36: warning: unused variable 'event' [-Wunused-variable]
   wifi_event_ap_stadisconnected_t* event = (wifi_event_ap_stadisconnected_t*) event_data;
                                    ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:346:36: warning: unused variable 'event' [-Wunused-variable]
      ip_event_ap_staipassigned_t * event = (ip_event_ap_staipassigned_t*)event_data;
                                    ^~~~~
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:356:23: warning: unused variable 'eth_handle' [-Wunused-variable]
      esp_eth_handle_t eth_handle = *(esp_eth_handle_t *)event_data;
                       ^~~~~~~~~~
[... 34 lines cut ...]
     ^
C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp:99:5: error: missing initializer for member 'tone_msg_t::frequency' [-Werror=missing-field-initializers]
[... cut ...]
-------------  --------  ------------
esp32dev       FAILED    00:00:26.761
==================================================================== 1 failed, 0 succeeded in 00:00:26.761 ==================================================================== 
Der Terminalprozess "C:\Users\Frank\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32dev'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.
```

hasenradball, 2022-03-31:
Are you sure this is the right version?
[image]

My first entry was with the version:
[image]

2.0.3 vs. 3.5.0

It would be nice to know whats the latest and greatest version of the core :-)

me-no-dev, 2022-04-01:
I think those 3.x.x versions are something that PIO guys did. It's not our version number. But your 2.0.3 seems to be the one that is latest

Jason2866, 2022-04-01:
I did not like the versioning numbers from pio (3.5.0 is Arduino core 1.0.6).
So i changed in my fork to be align with the Arduino core version. So 2.0.3 is latest.
The errors you have are normal. You have to adopt your code to the changes in Arduino core 2.0.3. Happy Coding :-)

hasenradball, 2022-04-01:
Thanks it was a little confusing for me with the numbers.
Schould I make some PR to prevent these warings?

hasenradball, 2022-04-01:
How Do I get the info if there is an new core Version?
Is then loading the esp32 core via Pio library manager the wrong way in general for the esp32?

Jason2866, 2022-04-01:
Lets hope the war in the Ukraine is soon over. Since Platformio is a Ukraine project.
When #6501 is merged you can use the released versions from here.
(Just do with `platform_packages = framework-arduinoespressif32 @ url to release zip`

closed 2025-01-06 by rftafas