espressif/arduino-esp32 #5453: ESP32 restart when client connecting

vidalouiswang, 2021-07-26:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
Sometimes, the ESP32 will rebstart when a device connecting to its soft AP.
And sometimes it will not happened.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include "BluetoothSerial.h"
#include<EEPROM.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <Update.h>

#define S(x,n)  (((x)<<n)|(x)>>(32-n))  //SHA定义S函数为循环左移

WebServer server(80);
BluetoothSerial SerialBT;
IPAddress apIP(192, 168, 1, 1);

bool inota = false;
String d = "";
int base = 0;
String lastTime = "";
int validValue = HIGH;
unsigned long h[5];
unsigned long m[16];
unsigned long w[80];

char* serverIndex = "<form method='POST' action='#' enctype='multipart/form-data' id='uploadForm'><input type='file' name='update'><input id='upload' type='button' value='Update'></form><script>get=e=>{return document.getElementById(e)};get('upload').onclick = function() {let x = new XMLHttpRequest(); x.open('post', '/update');x.upload.onprogress = e=> {this.value = (e.loaded / e.total * 100).toFixed(2)  + '%';};x.send(new FormData(get('uploadForm')));};</script>";

void sha1_pad(char *input, int len)
{
  int i;
  int n;
  for(i=0;i<16;i++){
    m[i] = 0;
  }
 
  for(i=0;i<len;i++){
    n = 24-((i&0x03)<<3);
    m[i/4] |= input[i]<<n;
  }
  n = 24-((i&0x03)<<3);
  m[i/4] |= 0x80<<n;
  m[15] = len*8;
}

[... 311 lines cut ...]
```
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x8011d049  A1      : 0x3ffd5c00  
A2      : 0x3ffe1e30  A3      : 0x3ffe1660  A4      : 0x3ffe1ae4  A5      : 0x3ffe1ac4  
A6      : 0x0201a8c0  A7      : 0x0c01a8c0  A8      : 0x8011cee8  A9      : 0x3ffd5bc0  
A10     : 0x3ffe1e40  A11     : 0x3ffe1660  A12     : 0x3ffd5c0c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffd5c00 0x4011d046:0x3ffd5c40 0x40129e5d:0x3ffd5c60 0x4012eed5:0x3ffd5ca0 0x401341be:0x3ffd5cc0 0x4011d977:0x3ffd5ce0 0x4008e689:0x3ffd5d10

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
```

Jason2866, 2021-07-26:
Triied already with a actual core version?

vidalouiswang, 2021-07-26:
> Triied already with a actual core version?
Hi,
I'm using the 1.0.4 core.

Jason2866, 2021-07-26:
The actual latest stable is 1.0.6

vidalouiswang, 2021-07-26:
> The actual latest stable is 1.0.6

Yeah, I change the url in board manager and updated to 1.0.6, the issue gone.
Thank you.
Didn't expect there is 1.0.6 could be update before.

closed 2021-07-26 by vidalouiswang