espressif/arduino-esp32 #5454: [Question] Increase HTTPD_MAX_REQ_HDR_LEN?

rtrbt, 2021-07-26:
Hi,

Would it be possible to increase (CONFIG_)HTTPD_MAX_REQ_HDR_LEN? Currently it is set to 512, however in some use cases web browsers send longer headers. An example: I'm using the Espressif HTTP server with digest authentication. The authorization header itself is 267 bytes long, all headers together around 550 bytes.

Bumping the max header length to e.g. 768 or 1024 would only increase the RAM usage if the Espressif HTTP server is used and only by the configured amount: HTTPD_MAX_REQ_HDR_LEN is only used for the [HTTPD_SCRATCH_BUF](https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/esp_http_server/src/esp_httpd_priv.h#L31) that is allocated only once as far as I have seen it in the code.

Some other users had the same problem in the past: #2983

I'm currently compiling the libs myself with the [lib-builder](https://github.com/espressif/esp32-arduino-lib-builder) but this seems to be broken at the moment (#5435).

VojtechBartoska, 2022-03-30:
Hello @rtrbt , we will take a look on this.

rtrbt, 2022-03-31:
Thanks, but this is already done: espressif/esp32-arduino-lib-builder@5a8939e9

closed 2022-03-31 by rtrbt

JoercT, 2022-04-05:
Hi.... how do I use this on windows please...