espressif/arduino-esp32 #5455: [E][WiFiUdp.cpp:183] endPacket(): could not send data: 12

francescot92, 2021-07-26:
Hi guys.
For my project I use Esp32 W-ROVER IE and Platformio to compiling the sketch.
PLATFORM: Espressif 32 (3.2.1) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 48 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32AnalogRead> 0.1.1
|-- <ESP32 Async UDP> 1.0.0
|-- <WiFi> 1.0
|-- <ArduinoJson> 6.17.1
|-- <UniversalTelegramBot> 1.3.0
|   |-- <ArduinoJson> 6.17.1
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <WiFiManager> 2.0.4-beta
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFi> 1.0
Building in release mode
Retrieving maximum program size .pio\build\esp-wrover-kit\firmware.elf
Checking size .pio\build\esp-wrover-kit\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.2% (used 43376 bytes from 327680 bytes)
Flash: [===       ]  30.4% (used 956882 bytes from 3145728 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM13
Uploading .pio\build\esp-wrover-kit\firmware.bin
esptool.py v3.0
Serial port COM13
Connecting......
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:4d:a7:b4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash params set to 0x0230

I wish to send a data to a localHost server. For this reasons i use the WiFiUdp library.
Sometimes in SerialTerminal i see these errors:

[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12

this error is a relative error because when the udp packet is not sent i try again in order to the packet will not be lost but, when this error is a ricorsive error causes the CORRUP HEAP and consequently the ESP32 restart:

[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
CORRUPT HEAP: Bad tail at 0x3fa5b704. Expected 0xbaad5678 got 0xa9836dd5
abort() was called at PC 0x400899e1 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008d500:0x3ffd9160 0x4008d779:0x3ffd9180 0x400899e1:0x3ffd91a0 0x40089b0d:0x3ffd91d0 0x401067d7:0x3ffd91f0 0x401028a1:0x3ffd94b0 0x40102829:0x3ffd9500 0x400922dd:0x3ffd9530 0x4008266e:0x3ffd9550 0x400d4064:0x3ffd9570 0x4008e7be:0x3ffd95c0

Rebooting...

how can i do? Why does this happen?

Thanks

Telboy99, 2021-08-19:
Did you get anywhere with this ? 

I have the exact same error testing the 'EspalexaBasic' on ESP32.

VojtechBartoska, 2022-03-30:
Hello @francescot92, sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska

elklaaso, 2023-12-08:
see https://github.com/espressif/arduino-esp32/issues/845
 the error errno 12 is ENOMEM is correct in my case, i freed up memory by moving esp32 sqlite to psram, and the error dissapeared. the error came after i made a large query that took a lot of memory to compile