espressif/arduino-esp32 #5459: WiFi.softAPdisconnect(true) always return false

KhanhNguyen199, 2021-07-26:
### Hardware:
Board:							ESP-WROOM-32
Core Installation version:			2.0.0-rc1
IDE name:						Arduino IDE
Flash Frequency:					240Mhz
PSRAM enabled:                                    NO
Computer OS:                                        Windows 10

### Description:
After creating the Wifi softapp, user cannot close softwap by using software.
Please check my code

//Change the code below by your sketch
#include <WiFi.h>

void setup() 
{
  Serial.begin(115200);
  Serial.println("Ready");

  WiFi.mode(WIFI_MODE_AP);
}

void loop() 
{
  if(true == WiFi.softAP("SoftAP test", "12345678", 8, 0))
  {
    Serial.println("Create softAp OK");
  }
  else
  {
    Serial.println("Create softAp NG");
  }

  delay(10000);

  if(true == WiFi.softAPdisconnect(true))
  {
    Serial.println("Close softAp OK");
  }
  else
  {
    Serial.println("Close softAp NG");
  }

  while(1);
}

### Debug Messages:
[2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Ready
[    47][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   135][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 10 - AP_START
Create softAp OK
[   634][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 11 - AP_STOP
[   634][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 10 - AP_START
[ 10644][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 11 - AP_STOP
[ 10647][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 10 - AP_START
E (21470) wifi_init_default: esp_wifi_get_mac failed with 12289
Close softAp NG[ 10653][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 11 - AP_STOP

Rotzbua, 2021-08-05:
I also observed this problem. Seems still present in master (https://github.com/espressif/arduino-esp32/commit/5bb8177aa1feec665938be97607b526a4f0ec0bb)
Error was `wifi_init_default: esp_wifi_get_mac failed with 12289`
No problem with `WiFi.softAPdisconnect(false)`

KhanhNguyen199, 2021-08-12:
Hello Rotzbua,

Do you have plan to fix this problem?
With current version, we cannot close the softAP by code.

Thanks!

Rotzbua, 2021-08-13:
@KhanhNguyen199 nope, not planned, sry

VojtechBartoska, 2022-03-30:
Hello, sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska