espressif/arduino-esp32 #5460: ESP-C3 internal temp sensor readings are incorrect with actual master

Jason2866, 2021-07-26:
It seems to be an issue between the ardiuino core and the idf code. On first access it reports fine. Next calls mess with a reinit of the sensor resulting in bad readings and this message: `E (312868) tsens: Do not configure the temp sensor when it's running!`
`E (312872) tsens: Temperature sensor is already running or not be configured`

What you are trying to do?
Get the temperature from internal temp sensor

Describe your system( Hardware, computer, O/S, core version, environment).
Windows 10 (latest), VSC/Platformio (latest), ESP32-C3 devkit running [Tasmota](https://github.com/arendst/Tasmota)

Describe what is failing.
Temp readings are fluctuating and wrong

### Hardware:
Board:							ESP32-C3 dev module
Core Installation version:			actual master 
IDE name:						Platform.io 5.1 
Flash Frequency:					80 Mhz
PSRAM enabled:                                  no
Upload Speed:					115200
Computer OS:                                       Windows 10

### Description:
Temp readings are wrong from internal sensor. Worked with 2.0.0 alpha

lbernstone, 2021-07-28:
The code is pretty simple.  Can you try directly calling the IDF functions and see if you get the same error?
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L64-L72

Jason2866, 2021-07-28:
It worked with alpha 2.0.0 core. The IDF code changed for the temp sensor and now it is broken.
I opened the issue to get a fix for the Arduino ESP32 part and not to do a workaround by calling IDF functions.

Jason2866, 2021-09-03:
Bump, the issue is also with the ESP32-S2 and core 2.0

VojtechBartoska, 2022-03-30:
Hello @Jason2866  , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? 

Thanks!

Jason2866, 2022-03-30:
@VojtechBartoska Thank you for the reminder, i will remove the workaround code and test.

VojtechBartoska, 2022-03-30:
Thanks @Jason2866, adding to Roadmap then and we will investigate it more. :)

PilnyTomas, 2022-04-06:
Hi @Jason2866, could you please provide a complete sketch that I can flash to replicate your issue?

Jason2866, 2022-04-06:
@PilnyTomas This is the code snipplet (used in Tasmota) for temp sensor reading
https://github.com/arendst/Tasmota/blob/development/tasmota/support_esp.ino#L519-L540

PilnyTomas, 2022-04-06:
Hi, I tested your code on S2 and it works perfectly.

Jason2866, 2022-04-08:
@PilnyTomas It works with the S2. The problem is still there with C3.

Jason2866, 2022-04-09:
@PilnyTomas Just did a Arduino framework with [this ](https://github.com/espressif/esp-idf/pull/8117)change in IDF https://github.com/espressif/esp-idf/pull/8117/files
With this the C3 temp sensor is working without a issue in Arduino.

VojtechBartoska, 2022-04-11:
@PilnyTomas Can you please confirmed this as @Jason2866 mentioned above? If yes, we can close this issue.

PilnyTomas, 2022-04-19:
Hi, same here - with the latest Arduino-esp32 core the temperature works on C3.

Jason2866, 2022-04-19:
Still have the issue as long not using https://github.com/espressif/esp-idf/pull/8117 
in IDF. 
Feel free to close, i found a solution to solve the issue

closed 2022-04-19 by PilnyTomas