espressif/arduino-esp32 #5461: How to override sdkconfig #defines ?

fmuntean, 2021-07-26:
I am using Platform.io
How can I overwrite #defines that are defined under the sdkconfig file? the generated sdkconfig.h does not contains failsafe #ifndef
to allow to overwrite it at the compile time using -D 
Please advise.

Jason2866, 2021-07-27:
You can not change settings from sdkconfig.h. This settings are "burned in" when compiling the ardunio libs is done. The sdkconfig.h file is for reference which options are set. If you want to change you have to build your own libs. This can be done with https://github.com/espressif/esp32-arduino-lib-builder
You can do the changes you want in the sdkconfig files and the libs are builded with this options.

fmuntean, 2021-07-27:
Thank you for the explanation.

closed 2021-07-27 by fmuntean