espressif/arduino-esp32 #5462: timerRead() returns "previous" timer value

g3gg0, 2021-07-26:
### Hardware:
Board:							Heltec WiFi LoRa v2
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Computer OS:                                        Windows 10

### Description:

I am trying to get exact timing information on a Heltec WiFi LoRa v2 and FSK modulated signals.
For accuracy reasons i switched to timerBegin() / timerRead().
After switching I realized that my channel hopping code was not working properly anymore.

Did a lot of debugging and found out that timerRead() always returns the **previous** timer value, the value that i expected from the previous call.

Executing timerRead() twice (!) gives the exact timer value i expected.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

void setup()
{
  Serial.begin(115200);
  
  hw_timer_t *timer_distance = timerBegin(2, 80, true);

  uint64_t timestamp1 = timerRead(timer_distance);
  delay(111);
  uint64_t timestamp2 = timerRead(timer_distance);
  delay(555);
  uint64_t timestamp3 = timerRead(timer_distance);
  delay(457);
  uint64_t timestamp4 = timerRead(timer_distance);
  uint64_t timestamp5 = timerRead(timer_distance);

  Serial.printf("1 to 2: %6u - expected 111000\n", timestamp2 - timestamp1);
  Serial.printf("2 to 3: %6u - expected 555000\n", timestamp3 - timestamp2);
  Serial.printf("3 to 4: %6u - expected 457000\n", timestamp4 - timestamp3);
  Serial.printf("4 to 5: %6u - expected      0\n", timestamp5 - timestamp4);
}

void loop()
{
}
```

### Messages:
```
1 to 2:     14 - expected 111000
2 to 3: 110841 - expected 555000
3 to 4: 555000 - expected 457000
4 to 5: 457000 - expected      0

```

g3gg0, 2021-07-28:
Hi Jason2866,

no didn't want to break other projects by updating the core.
However I did check the sources and [esp32-hal-timer.c](https://github.com/espressif/arduino-esp32/blob/8f46bade7afe7b38f861b9da8c7284b3988dd7b9/cores/esp32/esp32-hal-timer.c#L122) has the same code.

As I expected it to be a issue like "oh wait, before reading registers that depend on writing an other one, 
we would have to drain the write buffers by first reading the written back again" or similar.

Or in other words, i expected it to look like:
```
uint64_t timerRead(hw_timer_t *timer){
    timer->dev->update = 1;
    volatile uint32_t dummy = timer->dev->update; // make sure previous write happened before fetching
    uint64_t h = timer->dev->cnt_high;
    uint64_t l = timer->dev->cnt_low;
    return (h << 32) | l;
}
```

But I did neither try this, nor am I the right one to judge - maybe its something totally different :)

g3gg0, 2021-08-28:
any update on this?

me-no-dev, 2021-08-30:
Fixed in: https://github.com/espressif/arduino-esp32/pull/5498

closed 2021-08-30 by me-no-dev

g3gg0, 2021-08-30:
ah, thanks for the fix