espressif/arduino-esp32 #5463: Esp32C3 Serial1 can not use

lybgo, 2021-07-27:
### Hardware:
Board:							?ESP32C3 Dev Module? 
Core Installation version:			git master newest code
IDE name:						?Arduino IDE?
Flash Frequency:					?160Mhz?
Upload Speed:					?921600?
Computer OS:                                        ?Windows 10?

### Description:
When I use "Serial.begin(9600);" ,ESP32C3 reboot forever.
### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup()
{
  Serial.begin(115200);
  Serial1.begin(9600);
}

void loop()
{
}
```

### Debug Messages:
```
[12:58:06.227]收←◆ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x142c
load:0x403ce000,len:0xbe8
load:0x403d0000,len:0x292c
SHA-256 comparison failed:
Calculated: 382e1807e70f49d75fc0f369b096b655db26caa90c3a4372222295c5c880921f
Expected: 128706947ac2b0450ec378f21c43b80093368c6751e585ae0d834a044d6cc793
Attempting to boot anyway...
entry 0x403ce000
I (53) boot: ESP-IDF v4.4-dev-2030-gd93887f9f-dirty 2nd stage bootloader
I (53) boot: compile time 01:17:11
I (54) boot: chip revision: 3
I (54) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (61) qio_mode: Enabling default flash chip QIO
I (65) boot.esp32c3: SPI Speed      : 80MHz
I (69) boot.esp32c3: SPI Mode       : QIO
I (72) boot.esp32c3: SPI Flash Size : 4MB
W (76) boot.esp32c3: PRO CPU has been reset by WDT.
I (81) boot: Enabling RNG early entropy source...
I (85) boot: Partition Table:
I (88) boot: ## Label            Usage          Type ST Offset   Length
I (94) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (101) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (107) boot:  2 app0             OTA app          00 10 00010000 00140000
I (114) boot:  3 app1             OTA app          00 11 00150000 00140000
I (120) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (127) boot: End of partition table
I (130) boot_comm: chip revision: 3, min. application chip revision: 0
I (137) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=08398h ( 33688) map
I (149) esp_image: segment 1: paddr=000183c0 vaddr=3fc8a600 size=01474h (  5236) load
I (152) esp_image: segment 2: paddr=0001983c vaddr=40380000 size=067dch ( 26588) load
I (164) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=20224h (131620) map
I (185) esp_image: segment 4: paddr=0004024c vaddr=403867dc size=03ce0h ( 15584) load
I (188) esp_image: segment 5: paddr=00043f34 vaddr=50000000 size=00010h (    16) load
I (192) boot: Loaded app from partition at offset 0x10000
I (194) boot: Disabling RNG early entropy source...

[12:58:07.349]收←◆ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x142c
load:0x403ce000,len:0xbe8
load:0x403d0000,len:0x292c
SHA-256 comparison failed:
Calculated: 382e1807e70f49d75fc0f369b096b655db26caa90c3a4372222295c5c880921f
Expected: 128706947ac2b0450ec378f21c43b80093368c6751e585ae0d834a044d6cc793
Attempting to boot anyway...
entry 0x403ce000
I (53) boot: ESP-IDF v4.4-dev-2030-gd93887f9f-dirty 2nd stage bootloader
I (53) boot: compile time 01:17:11
I (54) boot: chip revision: 3
I (54) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (61) qio_mode: Enabling default flash chip QIO
I (65) boot.esp32c3: SPI Speed      : 80MHz
I (69) boot.esp32c3: SPI Mode       : QIO
I (72) boot.esp32c3: SPI Flash Size : 4MB
W (76) boot.esp32c3: PRO CPU has been reset by WDT.
I (81) boot: Enabling RNG early entropy source...
I (85) boot: Partition Table:
I (88) boot: ## Label            Usage          Type ST Offset   Length
I (94) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (101) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (107) boot:  2 app0             OTA app          00 10 00010000 00140000
I (114) boot:  3 app1             OTA app          00 11 00150000 00140000
I (120) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (127) boot: End of partition table
I (130) boot_comm: chip revision: 3, min. application chip revision: 0
I (137) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=08398h ( 33688) map
I (149) esp_image: segment 1: paddr=000183c0 vaddr=3fc8a600 size=01474h (  5236) load
I (152) esp_image: segment 2: paddr=0001983c vaddr=40380000 size=067dch ( 26588) load
I (164) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=20224h (131620) map
I (185) esp_image: segment 4: paddr=0004024c vaddr=403867dc size=03ce0h ( 15584) load
I (188) esp_image: segment 5: paddr=00043f34 vaddr=50000000 size=00010h (    16) load
I (192) boot: Loaded app from partition at offset 0x10000
I (194) boot: Disabling RNG early entropy source...
```

Jason2866, 2021-07-27:
Flash mode looks strange. Looks like a mismatch between mode DIO and QIO.
Try mode DOUT

lybgo, 2021-07-28:
> Flash mode looks strange. Looks like a mismatch between mode DIO and QIO.
> Try mode DOUT
My test code is the following simple lines, can you tell me which code to add to solve?
```
void setup(){
  Serial.begin(115200);
  Serial1.begin(9600);
}
void loop(){
}
```

Jason2866, 2021-07-28:
In Arduino IDE menu there should be the option to select the flash mode. I do not use Arduino IDE so maybe it is somewhere else to select the flash mode

lybgo, 2021-07-28:
> In Arduino IDE menu there should be the option to select the flash mode. I do not use Arduino IDE so maybe it is somewhere else to select the flash mode

I've tried all the Flash modes (QIO,DIO,DOUT,QOUT), but it still doesn't work out.

atanisoft, 2021-07-29:
@lybgo Try passing in pin numbers to use for Serial1. The defaults being picked (TX:17, RX: 18) overlap with the on-board flash (at least TX does) and that is why you are seeing a consistent reboot.

Try:
`Serial1.begin(9600, SERIAL_8N1, 18, 19);`

Note that pins 18 and 19 are also used for USB-JTAG.

lybgo, 2021-08-02:
@atanisoft I have tried your method, C3 will not restart constantly, but Serial1 cannot communicate, I have tried all pins (0-9,18,19).

Eliott-17, 2021-08-11:
Same here,

Trying to use GPIO 2 & 3, nothing happend on the pins:

  Serial1.begin(115200, SERIAL_8N1, RX1, TX1);
  Serial1.println("BOOT");

Eliott-17, 2021-08-15:
For the moment the only solution that I have: use function from ESP IDF:

`#include "C:\xxx\esp-idf\components\driver\include\driver\uart.h"

void ESP32uart()
{
    const uart_port_t uart_num = ECHO_UART_PORT_NUM;
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_CTS_RTS,
       .rx_flow_ctrl_thresh = 122,
    };
    // Configure UART parameters
    ESP_ERROR_CHECK(uart_driver_install(ECHO_UART_PORT_NUM, BUF_SIZE * 2, 0, 0, NULL, 0));
    ESP_ERROR_CHECK(uart_param_config(ECHO_UART_PORT_NUM, &uart_config));
    ESP_ERROR_CHECK(uart_set_pin(ECHO_UART_PORT_NUM, 2, 3, -1, -1));
}

uart_write_bytes(ECHO_UART_PORT_NUM, &rxValue[0], 6);

`

lbernstone, 2021-08-15:
You should not use headers from an external idf in arduino.  `#include <driver/uart.h>` should work and can be trusted to not have missing or conflicting symbols.

SuGlider, 2021-08-15:
HardwareSerial is currently under refactoring.
This will solve the current issue, as well as other problems.

Please hold for the news.

SuGlider, 2021-08-16:
Guys, I have created a PR to solve this issues, as well as other Arduino UART issues.
I confirm that the PR solves C3 Serial1 issue.

IF you want to try it and comment on the PR, please use 
https://github.com/SuGlider/arduino-esp32/tree/uart_refactoring 
or PR #5549

closed 2021-08-23 by me-no-dev

calcut, 2021-09-09:
Do we know if this also affects Esp32s2?
I'm having a very similar issue (using 2.0.0)

QGB, 2022-05-16:
```
 

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403814ee
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4ab8657f6f5aa80b81c7dd649db3bd64a98999ed7c65acc5a2ab77ce80caf525
Expected: 6babbc7b9ea63e04e0fc89ec224f6268aa3f84392018f73ab33721f768ce2eb6
Attempting to boot anyway...
entry 0x403ce000
Hello world!
E (156) uart: uart_set_pin(673): tx_io_num error
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40384a83
file: "C:\Users\qgb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32\esp32-hal-uart.c" line 168
func: uartBegin
expression: uart_set_pin(uart_nr, txPin, rxPin, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE)

abort() was called at PC 0x40384a87 on core 0
Core  0 register dump:
MEPC    : 0x403818d0  RA      : 0x40384a92  SP      : 0x3fc93260  GP      : 0x3fc8b400
TP      : 0x3fc89ad8  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x3fc9328c  S1      : 0x3fc9328c  A0      : 0x3fc93298  A1      : 0x3fc9327a
A2      : 0x00000000  A3      : 0x3fc932c5  A4      : 0x00000001  A5      : 0x3fc8d000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000001  S3      : 0x3fc8adb0
S4      : 0x0000000c  S5      : 0x00000000  S6      : 0x0000001c  S7      : 0x0000001b
S8      : 0x00000100  S9      : 0x00000000  S10     : 0x3fc8adb0  S11     : 0x0001c200
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc93260: 0x3fc8adb0 0x00000000 0x3fc93278 0x40389656 0x0000001c 0x00000000 0x00000030 0x3fc8bb14
3fc93280: 0x3fc9328c 0x3fc8bb30 0x3fc93278 0x38333034 0x37386134 0x40384a00 0x726f6261 0x20292874
3fc932a0: 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x34383330 0x20373861 0x63206e6f
3fc932c0: 0x2065726f 0x00000030 0x00000000 0x98a97d00 0x3c030a58 0x3fc8ada4 0x00000001 0x40384a8a
3fc932e0: 0x00000001 0x3fc8ada4 0x00000001 0x42001bb6 0x3c0305f4 0x0001c200 0x00000003 0x00000000
3fc93300: 0x00000001 0x00000000 0x3fc8e170 0x00000000 0x00000000 0xffc1ffff 0x00000000 0x00000000
3fc93320: 0x00000000 0x00000000 0x00004e20 0x00000001 0x00000070 0x00000000 0x0800001c 0x0001c200
3fc93340: 0x0000001c 0x0000001b 0x3fc8c1fc 0x42000cc6 0x00000070 0x00000000 0x00000000 0x00000000
3fc93360: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93380: 0x00000000 0x0001c000 0x3fc8c000 0x420000d0 0x00000000 0x00000000 0x00000000 0x00000000
3fc933a0: 0x00000000 0x3fc8d000 0x42001000 0x42001512 0x00000000 0x00000000 0x00000000 0x40386fe8
3fc933c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc933e0: 0x27146414 0x01010107 0x00000001 0x00000000 0x00000000 0x00000000 0xffffffff 0x7fefffff
3fc93400: 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3fe00000 0x00000000 0x3ff80000
3fc93420: 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000 0x3ff00000
3fc93440: 0x00000000 0x40240000 0x00000000 0x401c0000 0x00000000 0x40140000 0x00000000 0x43500000
3fc93460: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fc93190 0x3fc8d264 0x3fc8cbec
3fc93480: 0x3fc8cbec 0x3fc93474 0x3fc8cb
──────────────────────────────────────────────

Session stopped
    - Press <return> to exit tab
    - Press R to restart session
    - Press S to save terminal output to file

Error reading from serial device

```

lbernstone, 2022-05-16:
Please don't hijack issues.  This issue was for a previous version.  Open a new issue and follow the posting instructions.

PR #5549 (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF
## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.
commit b1d072df (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF (#5549)

## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.