espressif/arduino-esp32 #5464: WebServer Arduino Board Library 1.0.6 do not function anymore: 1.0.4 was OK.

rocrail, 2021-07-27:
The function 
`virtual WebClient client()`
always returns NULL after a client did connect.
So the WebServer is no longer compatible with 1.0.4.
Please fix this.

VojtechBartoska, 2021-08-18:
@rocrail Hi, can you please try this with latest 2.0.0 rc1?

rocrail, 2021-08-20:
Sorry, no cure.

rocrail, 2022-02-01:
Please fix this incompatibility issue.
2.0.0 rc1 does not fix it.
For the CAM I have to stick with 1.0.4, but my application must switch from SPIFFS to LittleFS which is not available in 1.0.4...

me-no-dev, 2022-02-01:
what is that WebServer and WebClient library? Could you provide any code that can help replicate the issue?

VojtechBartoska, 2022-04-07:
Closing as Expired, if needed, please open new issue @rocrail. Thanks.

closed 2022-04-07 by VojtechBartoska