espressif/arduino-esp32 #5467: Esp32-c3 digitalRead(pin) always return 0 "LOW"

vanderareosa, 2021-07-27:
Trying  to get the status of the pin using digitalRead(pin)
Hardware:
MAC OS 10.13
Arduino IDE 1.8
board: ESP32-C3-DevKitM-1

when I trying to get the status of pin:
int status = digitalRead(pin)
or 
digitalWrite(pin, !digitalRead(pin))
Always the return is 0 "LOW" even the pin being HIGH

chegewara, 2021-07-27:
1) did you set `pinMode`?
2) could you try to call `gpio_reset_pin` before `pinMode`?
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#_CPPv414gpio_reset_pin10gpio_num_t

vanderareosa, 2021-07-27:
I'm a little confused by this documentation you sent the link to, before I define the pin: #define relayPin 4
set the mode: pinMode(relayPin, OUTPUT)
and changed the status digitalWrite(relayPin, !(digitalRead(relayPin)). Did everything change? Because now I got an error setting the pin using #define "invalid conversion from 'int' to gpio_num_t. How do I define a pin now? how change status? and how do i set pinMode? Thanks for your attention.

chegewara, 2021-07-27:
`digitalWrite(relayPin, !(digitalRead(relayPin))`
This will not work. You cant read value if pin is OUTPUT, you have to use variable for that to keep it status.

vanderareosa, 2021-07-27:
this is the sequence of my code:
#define RelayPin 4  
pinMode(RelayPin, OUTPUT);
int status  = digitalRead(RelayPin);
if (status==0)
   digitalWrite(RelayPin, HIGH);
else
   digitalWrite(RelayPin, LOW);

but always digitalRead(RelayPin); return 0
I tried to do that in ESP32-WROOM and worked perfectly, but in esp32-C3 not. Even digitalWrite(relay Pin, !(digitalRead(relay Pin)) worked in ESP32-WROOM

chegewara, 2021-07-27:
Try this before `pinMode`:
`gpio_reset_pin((gpio_num_t) RelayPin);`

vanderareosa, 2021-07-27:
gpio_reset_pin((gpio_num_t) RelayPin);  
  pinMode(RelayPin, OUTPUT);

didn't works too, still return only zero. I tried to change the pin and nothing.
Do you have some simple example of like blink to ESP32-C3?
I think C3 is very new so, I didn't find any example in the internet. But just C3 suport Firebase and BLE in same time in my project.

vanderareosa, 2021-07-27:
I tried to do as the example below but it didn't work either.

#define GPIO_BIT_MASK  (1ULL<<RelayPin) 
#define RelayPin 4

void setup() {
  Serial.begin(115200);
  
  gpio_reset_pin((gpio_num_t) RelayPin);
  
  gpio_config_t io_conf;
  io_conf.intr_type = GPIO_INTR_DISABLE;
  io_conf.mode = GPIO_MODE_OUTPUT;
  io_conf.pin_bit_mask = GPIO_BIT_MASK;
  io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
  io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
  gpio_config(&io_conf);  
}

void loop() {

  delay(5000);
  int relayStatus = gpio_get_level((gpio_num_t)RelayPin);
  if (relayStatus==0){
    Serial.println("Will turn on relay");
    gpio_set_level((gpio_num_t)RelayPin, 1);
  }else{
    Serial.println("Will turn off relay");
    gpio_set_level((gpio_num_t)RelayPin, 0);
  }
    
}

atanisoft, 2021-07-29:
@vanderareosa gpio_get_level (and likely digitalRead) will return LOW (0) for any OUTPUT only pin. If you want to read the state try INPUT_OUTPUT mode which is available in IDF only it looks like.

vanderareosa, 2021-07-29:
@atanisoft worked.. thanks so much.. you save my week.. :)

closed 2021-07-29 by vanderareosa