espressif/arduino-esp32 #5470: ESP32-S2 I2C getting stuck in write loop after first ACK

caternuson, 2021-07-27:
### Hardware:
Board:							[Adafruit FunHouse](https://www.adafruit.com/product/4985)
Core Installation version:			2.0.0-rc1
IDE name:						Arduino IDE 1.8.13
Flash Frequency:					80Mhz
PSRAM enabled:                                    yes
Upload Speed:					921600
Computer OS:                                        Ubuntu

### Description:
Can not initialize the included I2C sensors (DPS310 and AHT20). Attempting an [I2C scan](https://playground.arduino.cc/Main/I2cScanner/) results in board reset cycling.

### Sketch:
You can probably see this by using any ESP32-S2 board, wiring up something to I2C, and running the [I2C scanner sketch](https://playground.arduino.cc/Main/I2cScanner/).

Example shown here is for attempting to directly use the DPS310 sensor, which is at `0x77`. [Library here](https://github.com/adafruit/Adafruit_DPS310/).
```cpp
#include <Adafruit_DPS310.h>

Adafruit_DPS310 dps;

void setup() {
  Serial.begin(9600);
  while (!Serial) delay(10);
  Serial.println("DPS310 Test.");
  
  if (! dps.begin_I2C()) {
    Serial.println("Failed to find DPS");
    while (1) yield();
  }
  Serial.println("DPS OK!");
}

void loop() {
}
```

### Debug Messages:
This is a run time issue. Code compiles and uploads fine. Serial monitor output FWIW:
[image]

Here is what is happening on the I2C bus. A write is initiated and the sensor ACKs:
[image]

Then it goes into an infinite loop writing out `0x00`. 
[image]

Those `0x00` writes only stop when the board finally resets at some point.

marcboon, 2021-08-11:
I'm using a SSD1306 OLED display on a ESP32S2-Saola-1 board, with the same 2.0.0-rc1 core version, and the I2C works fine, using the Adafruit SSD1306 library.

chegewara, 2021-08-11:
After checking this board schematics and Wire library i would say it is expected it wont work. What is interesting is that you are getting anything with logic analyzer.
That board is using pins 33 and 34 for I2C which are not standard I2C pins.
Another interesting thing is this adafruit sef-test code which is the same and seems it should work:
https://learn.adafruit.com/adafruit-funhouse/arduino-self-test-example

My suggestion is to: 
```
Wire.begin(34, 33);
...
dps.begin_I2C(0x77, &Wire);
```

caternuson, 2021-08-11:
Wouldn't it end up using the pins defined in the variants file:
https://github.com/espressif/arduino-esp32/blob/a4118ea88987c28aac3a49bcb9cc5d6c0acc6f3f/variants/adafruit_funhouse_esp32s2/pins_arduino.h#L42-L43
when it gets here for SDA:
https://github.com/espressif/arduino-esp32/blob/a4118ea88987c28aac3a49bcb9cc5d6c0acc6f3f/libraries/Wire/src/Wire.cpp#L77
and here for SCL:
https://github.com/espressif/arduino-esp32/blob/a4118ea88987c28aac3a49bcb9cc5d6c0acc6f3f/libraries/Wire/src/Wire.cpp#L94

chegewara, 2021-08-11:
@caternuson Yes, as long as you pick the correct board variant.

caternuson, 2021-08-11:
The Arduino IDE was set to Tools -> Board -> ESP32 Arduino -> Adafruit FunHouse. Is there an additional step?

chegewara, 2021-08-11:
That should be enough, but just in case i would test i2c_scanner with explicit pins in `Wire.begin(34, 33)`.
Im not saying it will fix your problem, but who knows.

caternuson, 2021-08-11:
This I2C scanner sketch:
```cpp
#include <Wire.h>

void setup()
{
  Wire.begin(34, 33);
  Serial.begin(9600);
  while (!Serial);
  Serial.println("\nI2C Scanner");
}

void loop()
{
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknown error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}
```
Results in this in the Serial Monitor:
[image]

And this on the I2C bus:
[image]

Once it gets to the first address with a device and gets an `ACK`, it gets stuck in same loop mentioned in first post above. 

Here's the last address (0x37) ***without*** a device:
[image]

And the next address (0x38) which does have a device:
[image]

The write sequence cutoff in that capture continues forever:
[image]

chegewara, 2021-08-11:
@caternuson I dont have this board, so my question:
- does this board use native USB? 
After selecting the board in arduino IDE i can see it is enabling USB CDC, so i am guessing it does.

I did some test with custom board that is using native USB and I2C on pins 16/17 works fine, then i changed I2C pins to 33/34 and code stuck. Next test i performed with saola and both tests passed.

@me-no-dev I am suspecting some conflict between I2C on particular pins and native USB.

@caternuson Could you perform one more test (it may be not easy):
- please change `USB CDC enable on boot => disable`
OR 
- Serial over `UART`
i dont know which one option exists in RC1 
You will not see any logs, but you should see change with logic analyzer. Thanks

caternuson, 2021-08-11:
@chegewara Thanks for looking at this. Yep, it's native USB. Here's what the Arduino IDE looks like with RC1:
[image]

Sorry, not sure I follow your suggested next test. Tried switching to the UART0 option, but seeing the same results in the logic analyzer. Were you suggesting something else? Happy to test whatever. Also, FWIW, the only other ESP32-S2 board I currently have access to is an Unexpected Maker FeatherS2.

chegewara, 2021-08-11:
@caternuson It is bug in arduino-esp32 and has to be fixed

caternuson, 2021-08-11:
@chegewara Thanks. Need any more info or tests from me?

chegewara, 2021-08-11:
@me-no-dev I think this is breaking PR, but didnt confirmed yet:
https://github.com/espressif/arduino-esp32/commit/5502879a5b25e5fff84a7058f448be481c0a1f73#

chegewara, 2021-08-11:
@caternuson There is simple temporary fix for the issue that allows to run scan and probably I2C too.
Try this fix on your local arduino-esp32:
```
i2c_err_t i2cWrite(i2c_t * i2c, uint16_t address, uint8_t* buff, uint16_t size, bool sendStop, uint16_t timeOutMillis){
	esp_err_t ret = ESP_OK;
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, (address << 1) | I2C_MASTER_WRITE, ACK_CHECK_EN);
    if(size) 
        i2c_master_write(cmd, buff, size, ACK_CHECK_EN);
    //if send stop?
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c->num, cmd, timeOutMillis / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    return ret;
}
```
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.c#L1846-L1857

caternuson, 2021-08-12:
@chegewara That seems to work. I tweaked the scanner sketch slightly to have the begin call just be:
```cpp
Wire.begin();
```
just to test that the variant info is working as expected as well.

With your BSP changes above, now get this (expected) output in Serial Monitor:
[image]

Logic trace is also as expected:
[image]

me-no-dev, 2021-08-12:
`if(size)` is the change? @chegewara

chegewara, 2021-08-12:
> `if(size)` is the change? @chegewara

yes

me-no-dev, 2021-08-12:
at least it fixes scan :) I would have thought that IDF would handle this situation...

me-no-dev, 2021-08-12:
@chegewara want to add this fix as PR?

chegewara, 2021-08-12:
Thats possible. I tested IDF with this example just to perform scan:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2c/i2c_tools

Since the fail is when IDF function is performing write with 0 length buffer it may be issue with IDF.

> @chegewara want to add this fix as PR?

no, thanks but no. i feel offended by espressif, so no more PRs

SuGlider, 2021-09-14:
@caternuson 
Please check Arduino Core 2.0.0 PR #5664 
This is a complete I2C Wire refactorng. Success reported on #4729 and on the PR itself #5664.
It may solve the issue as well.

caternuson, 2021-09-14:
@SuGlider Is that an alternative to #5528? That's been merged and fixed this issue as well.

SuGlider, 2021-09-14:
@caternuson 
It is the whole fix, including i2c scan and a better way to send reStart and other i2c events.

caternuson, 2021-09-15:
@SuGlider Tested #5664 and it works with an I2C scanner and the DPS310 test sketch if original post above.

Just to make sure I tested correctly, here are the general steps taken:
* Install ESP32 BSP 2.0.0 via Arduino IDE Boards Manager
* Cloned this repo `git clone ...` to a temporary location
* Fetched the PR code `git fetch origin pull/5664/head:pr5664`
* Checkout those changes `git checkout pr5664`
* Manually copied the 4 files into the appropriate location under `.arduino15/packages/esp32/hardware/esp32/2.0.0`
* Compiled/upload sketches via Arduino IDE

SuGlider, 2021-09-15:
Yes, perfect.
Happy that everything worked as expected.

Thanks for taking time to test it and report back here!

me-no-dev, 2021-09-20:
Hi could you please re-test with this new PR: https://github.com/espressif/arduino-esp32/pull/5683

caternuson, 2021-09-20:
@me-no-dev Tested #5683 on Funhouse with I2C scanner sketch and DP310 test sketch. Same basic tests as above. Both sketches ran as expected.

me-no-dev, 2021-09-21:
Thanks @caternuson !

PR #5683 (2021-10-01): Implement Thread-Safe I2C based on ESP-IDF API
## Summary
This PR is a complete refactoring of  Wire Library and I2C HAL in order to use IDF instead of current Register manipulation approach.

It implements Thread-safe Arduino Wire.

Fix #5478 
Fix #5635 
Fix #5636 
Fix #4729 
Fix #5470 

## Impact
It solves many reported issues related to I2C.
closed 2021-10-01 by me-no-dev