espressif/arduino-esp32 #5473: RX Full! Discarding

yuranikspb, 2021-07-28:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			latest from platform io
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:

when i run my sketch in simple mode inside loop() {}
 work good

when i am creat task with this code i have 
error after some times.
[E][BluetoothSerial.cpp:316] esp_spp_cb(): RX Full! Discarding 13 bytes

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-
```
xTaskCreatePinnedToCore(
      TaskComunicationCode, /* Task function. */
      "TaskComunication",   /* name of task. */
      10000,                /* Stack size of task */
      NULL,                 /* parameter of the task */
      5,                    /* priority of the task */
      &TaskComunication,    /* Task handle to keep track of created task */
      0);                   /* pin task to core 0 */

  //create a task that will be executed in the Task2code() function, with priority 1 and executed on core 1
  xTaskCreatePinnedToCore(
      TaskAnimationCode, /* Task function. */
      "TaskAnimation",   /* name of task. */
      10000,             /* Stack size of task */
      NULL,              /* parameter of the task */
      5,                 /* priority of the task */
      &TaskAnimation,    /* Task handle to keep track of created task */
      1);                /* pin task to core 1 */

void TaskComunicationCode(void *pvParameters)
{ 
#ifdef DEBUG_SERIAL
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());
#endif
  for (;;)
  {
esp_task_wdt_reset();
    while (SerialBT.available())
    {
      String rxValue = SerialBT.readStringUntil('\n');
      if (rxValue.length() > 0)
      {
#ifdef DEBUG_SERIAL
        Serial.print("Received value: "); //  "Полученное значение: "

        for (int i = 0; i < rxValue.length(); i++)
        {
          Serial.print(rxValue[i]);
        }
#endif

        if (rxValue.indexOf("ANM") != -1)
        {
#ifdef DEBUG_SERIAL
          Serial.print("ANM="); //  " - номер анимации получен
          Serial.print(rxValue.substring(3, 6).toInt());
          Serial.print("SPD="); //  " - Скорость анимации получена
          Serial.println(rxValue.substring(6, 9).toInt());
#endif
          PlayAnimation = true;
          selectedAnimation = rxValue.substring(3, 6).toInt();
          Animation_speed = rxValue.substring(6, 9).toInt();
        }

        if (rxValue.indexOf("CLR") != -1)
        {
#ifdef DEBUG_SERIAL
          Serial.print("CLR=R"); //  " - цвет получен
          Serial.print(rxValue.substring(3, 6).toInt());
          Serial.print("G"); //  " - цвет получен
          Serial.print(rxValue.substring(6, 9).toInt());
          Serial.print("B"); //  " - цвет получен
          Serial.println(rxValue.substring(9, 12).toInt());
#endif
          PlayAnimation = false;

          while(!Animation_stoped) //wait for animation stoped
          {
            ;
          }
          setAll(rxValue.substring(3, 6).toInt(), rxValue.substring(6, 9).toInt(), rxValue.substring(9, 12).toInt(), 0);
          ledcWrite(led_red_ch, 255 - rxValue.substring(3, 6).toInt());
          ledcWrite(led_green_ch, 255 - rxValue.substring(6, 9).toInt());
          ledcWrite(led_blue_ch, 255 - rxValue.substring(9, 12).toInt());
        }

        if (rxValue.indexOf("ON") != -1)
        {
#ifdef DEBUG_SERIAL
          Serial.println(" - LED ON"); //  " - светодиод включен"
#endif
          digitalWrite(ledPin, LOW);
        }
        else if (rxValue.indexOf("OFF") != -1)
        {
#ifdef DEBUG_SERIAL
          Serial.println(" - LED OFF"); //  " - светодиод выключен"
#endif
          digitalWrite(ledPin, HIGH);
        }
      }
         
    }
 
  }
}
```

SukhoiRU, 2021-09-20:
Hello! I've had the same problem. Under Arduino IDE works fine, but under VS Code with ESP-IDF and arduino libraries as component fails. Solution was using callback. Not register_callback, but onData. 

```
BluetoothSerial ESP_BT;
TinyGPSPlus     gps;

void  callback_ESP_BT_data(const uint8_t *buffer, size_t size)
{
    for(size_t i = 0; i < size; i++)
    {
      const char  c = buffer[i];
      gps.encode(c);
      //printf("%c", c);
    }
}

//Somewere in main

ESP_BT.onData(callback_ESP_BT_data);

```

italocjs, 2021-12-10:
> Hello! I've had the same problem. Under Arduino IDE works fine, but under VS Code with ESP-IDF and arduino libraries as component fails. Solution was using callback. Not register_callback, but [...]

This fixed it for me,  thanks!

VojtechBartoska, 2022-04-07:
Serial was refactored lately so this need to be tested on v2.0.3-RC1 or can we closed it be closed @yuranikspb?

yuranikspb, 2022-04-07:
I can't test it now, because work at another project. You can close this issue.

closed 2022-04-08 by VojtechBartoska