espressif/arduino-esp32 #5474: High memory consumption on 2.0.0rc1 core

ale-trevizoli, 2021-07-28:
Hi. 
I'm using esp32-dev board 4Mb Flash.
I have a system developed with Arduino IDE 1.8.5 and ESP32 core 1.0.6.
This system print on loop() - ESP.getFreeHeap() each 5 seconds, and show me ~302K free.
I exchange 1.0.6 to 2.0.0rc1 and I only compiled again and got ~265K free, difference of 37Kbytes.
What's can happening?

Thanks!

SuGlider, 2021-08-11:
I tested the  sketch below and got  351315 bytes free on heap.

'''
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println((unsigned long)ESP.getFreeHeap());
  delay(5000);
}

''''

avillacis, 2021-09-02:
This issue of additional memory use persists in recently-released 2.0.0 stable versus 1.0.6. In my case, the difference is enough to have 1.0.6 stashed alongside 2.0.0 in case one of our projects fails due to memory starvation.

For example, consider the following sketch:

[WiFiScan-ESP32S2.txt](https://github.com/espressif/arduino-esp32/files/7101947/WiFiScan-ESP32S2.txt)

This sketch compiles properly under 1.0.6 and 2.0.0. However, note the difference in memory use, especially after WiFi is set up:

Under 1.0.6:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
MEMORY USE AT setup() before WiFi setup: 360808 total, 334340 free, 113792 max.alloc
Setup done
MEMORY USE AT setup() after WiFi setup: 359568 total, 283652 free, 113792 max.alloc
scan start
scan done
12 networks found
(redacted)

MEMORY USE AT after WiFi scan: 359484 total, 282476 free, 113792 max.alloc
scan start
scan done
17 networks found
(redacted)

MEMORY USE AT after WiFi scan: 359484 total, 282076 free, 113792 max.alloc

```

Under 2.0.0:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
MEMORY USE AT setup() before WiFi setup: 366171 total, 308455 free, 65524 max.alloc
Setup done
MEMORY USE AT setup() after WiFi setup: 364203 total, 228051 free, 65524 max.alloc
scan start
scan done
14 networks found
(redacted)

MEMORY USE AT after WiFi scan: 364179 total, 226875 free, 65524 max.alloc
scan start
scan done
13 networks found
(redacted)

MEMORY USE AT after WiFi scan: 364179 total, 226955 free, 65524 max.alloc
```

By switching SDK releases and doing nothing else, the free memory has decreased from 282076 to 226955 (an unexplained additional 55121 bytes), and the maximum allocatable block has decreased from 113792 to 65524 bytes.

Why????

SuGlider, 2021-09-03:
1.0.6 uses IDF 3.3
2.0.0 uses a RC for IDF 4.4
WiFi is completely based on IDF.

I guess this is the reason for a higher RAM consumption.

SuGlider, 2021-09-03:
Maybe, in the near future, on the final release of IDF 4.4 and respective rebase of Arduino Core to this IDF, it may improve.

No promises....

me-no-dev, 2021-09-07:
there are many reasons why there is a different memory usage in recent IDF. This is in NO WAY connected to Arduino. Arduino's is mostly disabling some IRAM optimizations, because users have sketches with WiFi, BT and all kinds of other things that make compilation impossible. That change though actually gives you memory. Do you actually have memory issues?

avillacis, 2021-09-07:
> there are many reasons why there is a different memory usage in recent IDF. This is in NO WAY connected to Arduino. Arduino's is mostly disabling some IRAM optimizations, because users have sketches with WiFi, BT and all kinds of other things that make compilation impossible.

Maybe some additional IRAM optimization was overlooked?

> That change though actually gives you memory. Do you actually have memory issues?

Yes, I do have memory issues with some of my projects that were absent when compiled under 1.0.6. At least one of our projects (the firmware for the [OpenVenti artificial respirator project](https://openventi.org/en/)) is unable to complete its OTA firmware update procedure due to insufficient RAM under 2.0.0, where the same source code worked successfully with 1.0.6. This has forced me to look for and remove RAM inefficiencies in the process, but the fact remains that 55Kb extra RAM use is currently a showstopper for updating to 2.0.0, at least for this project.

igrr, 2021-09-07:
Here is some additional data based on IDF example, [examples/wifi/getting-started/station](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station):

| IDF release | sdkconfig | Heap [1] after app_main | Heap [1] after esp_wifi_start | Heap used |
|---|---|---|---|---|
| IDF master (v4.4-dev-2890-ge93cee4605) | from arduino-esp32 2.0.0 [2] | 326435 | 251415 | 75kB |
| IDF master (v4.4-dev-2890-ge93cee4605) | default | 323884 | 284048 | 40kB |
| IDF release/v3.3 (v3.3.4-404-g7e63061fae) | from arduino-esp32 1.0.6 [3] | 337720 | 291300 | 46kB |
| IDF release/v3.3 (v3.3.4-404-g7e63061fae) | default | 327232 | 290864 | 36kB |

First, note that with the default sdkconfig, IDF 4.4-dev has about 7kB less heap at app_main than IDF release/3.3. See the note [1] below, though. These numbers include "IRAM" heap; if we compare only DRAM heap (MALLOC_CAP_DEFAULT), then v4.4-dev has 2kB **more** heap than 3.3 (277032 bytes free in v4.4-dev, 275824 free in release/v3.3) when this example starts up — see the table below.

We can see that some RAM can be reclaimed by changing the configuration (sdkconfig) that arduino-esp32 is built with. The sdkconfig used in 2.0.0 results in 35kB higher RAM usage than the default sdkconfig (for the same IDF version). In 1.0.6 this was only 10kB higher.

For the rest of the difference between 1.0.6 and 2.0.0 heap usage we should look at the startup heap size: 334340 vs 308455 (26kB difference). This might be due to higher static memory usage (.data, .bss) and/or higher heap usage. The issue might be on IDF side or Arduino-esp32 side, it is hard to say without investigating it in detail. For the static memory usage, `idf.py size-components` (or idf_size.py) can provide per-library breakdown of static data — including libarduino.a. For the dynamic memory usage, we can enable [heap tracing](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-tracing) to find the allocations performed at startup (before `setup` is called). I haven't had time to look into this part yet, but I suspect that there is room for improvement.

---

**Note 1:** note that arduino-esp32 [ESP.getFreeHeap() function returns](https://github.com/espressif/arduino-esp32/blob/92ce408f4c71d235f0961a265e8d202f9fd570d5/cores/esp32/Esp.cpp#L125-L128) the amount of internal DRAM _and_ IRAM; the latter is not available via `malloc`, so the number is somewhat higher than the useful amount of heap memory. See https://github.com/espressif/arduino-esp32/issues/5346. To make comparisons easier, i used the same way of measuring the heap size in IDF examples, even though that's not an entirely accurate approach.

Here is the version of the table above, but this time showing only the DRAM heap (MALLOC_CAP_DEFAULT). Note that "Heap used" (difference) is the same as above, just the initial number in each row is less by the amount of free IRAM in the particular configuration:

| IDF release | sdkconfig | Heap after app_main | Heap after esp_wifi_start | Heap used |
|---|---|---|---|---|
| IDF master (v4.4-dev-2890-ge93cee4605) | from arduino-esp32 2.0.0 [2] | 239484 | 164040 | 75kB |
| IDF master (v4.4-dev-2890-ge93cee4605) | default | 277032 | 237432 | 40kB |
| IDF release/v3.3 (v3.3.4-404-g7e63061fae) | from arduino-esp32 1.0.6 [3] | 272068 | 225116 | 47kB |
| IDF release/v3.3 (v3.3.4-404-g7e63061fae) | default | 275824 | 239288 | 36kB |

**Note 2:** https://raw.githubusercontent.com/espressif/arduino-esp32/2.0.0/tools/sdk/esp32/sdkconfig
**Note 3:** https://raw.githubusercontent.com/espressif/arduino-esp32/1.0.6/tools/sdk/sdkconfig

chegewara, 2021-10-16:
`Reason: 2 - AUTH_EXPIRE`
This is normal when you are using smartphone as AP (maybe with routers too) during development. 
When esp32 is connected and you re-flash it then smartphone keeps this connection active, even if esp32 disconnected. Then after reset esp32 is trying to connect again, but smartphone still wants to update old connection.
I dont know how it is handled in low level wifi packets exchange, just saying from experience. 
There is 2 options that work for me:
1. during re-flash disable and re-enable AP,
2. or after re-flash esp32 and that error occured just restart esp32 with pwr button

olicooper, 2021-10-16:
Thanks for your reply,
The router I am using is actually an AC series Linksys which is only a few years old. I modified the settings to disable WPA3 just in-case that was causing the issue, but I will also try restarting the router because I haven't tried this yet. I erased the flash and then flashed Arduino 1.0.6 and it worked perfectly again, then I went back to the `mem-optimized` branch and observed the same connection issues. I will do more testing based on your comments and to confirm if it is working for the `main` branch.

SuGlider, 2021-10-16:
@olicooper,
Upon your GitHub, it says you are using IDF 4.3. Is it right?
The branch I sent uses 4.4 latest master.

Could you please confirm it by executing:
```
Serial.println("Using ESP object:"); Serial.println(ESP.getSdkVersion());
```

olicooper, 2021-10-17:
Yeah I kept noticing that static IP too, but I am not calling config in my code anywhere. As I said above, I have also tried the example `WiFiClientBasic.ino` code and only changing the ssid and password: https://github.com/espressif/arduino-esp32/blob/mem-optimized/libraries/WiFi/examples/WiFiClientBasic/WiFiClientBasic.ino
You can see from the logs it says about a static IP there too.

I call connect a few lines down:
https://github.com/olicooper/arduino-v2-esp32-platformio/blob/332caebd3ef3e8d4aa94241265bb1be710b1c4c1/src/main.cpp#L84

SuGlider, 2021-10-19:
@olicooper 
I have found the issue with WPA and WPA2.
It's now fixed on ```mem-oprimized``` branch.
Related commit: https://github.com/espressif/arduino-esp32/commit/f459b101f3bdf0ed0ece590eeb9f958ed818bd3e

commit 67c99142 (2021-10-21): Allows spiram malloc with wifi dynamic buffers - better free heap (#5791)

Summary

Modifies WiFi lib to allow dynamic buffer allocation along with SPIRAM MALLOC enabled
This gives more heap space to the users

Related PR in Arduino Lib Builder: espressif/esp32-arduino-lib-builder#47

Impact

WiFi will work the same as it was in version 1.0.6, restoring free heap.

close #5630
close #5474
close #5699
close #5697