espressif/arduino-esp32 #5476: 2.0.0-rc1 analogRead() incorrect readings

szerwi, 2021-07-29:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			2.0.0-rc1
IDE name:						Arduino IDE 2.0.0-beta9
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
```analogRead()``` gives incorrect readings on 2.0.0-rc1 comparing to 2.0.0-alpha1 and olders.

I am using ESP32 DevKitC V4 with external antenna connector. I have tested it on pins 36 (VP), 33 and 27.
Multimeter shows 2.52V
ESP on 2.0.0-alpha1 libraries returns ~3000, whereas 2.0.0-rc1 libraries return 4095.
It seems like 2.0.0-rc1 libraries map 0-1V to 0-4095 and everything above 1.0V is 4095.

### Sketch:
```cpp
#define pin 27

void setup() {
  Serial.begin(115200);
}

void loop() {
  int value = analogRead(pin);

  Serial.println(value);
  Serial.println("");
  delay(1000);
}

```

garageeks, 2021-08-01:
I have noticed wrong readings too compared to previous versions, will investigate it and report as well

Lackmann1994, 2021-08-04:
I think the default attenuation changed by using the latest ESP IDF
try setting the attenuation explicitly before analogRead(YOUR_GPIO):

analogSetPinAttenuation(YOUR_GPIO, adc_attenuation_t::ADC_11db);
or
analogSetAttenuation(adc_attenuation_t::ADC_11db);

szerwi, 2021-08-08:
@Lackmann1994 Unfortunately, it doesn't change anything in my case.

marcboon, 2021-08-08:
On a ESP32-S2-Saola-1 module, using 2.0.0-rc1, I get 13 bit results from ADC1: analogRead(1) returns 8191 for any voltage above 1V approx.
The reference voltage (full-scale seems to be about 1V, which corresponds to no attenuation (0dB).

> I think the default attenuation changed somehow.
> I think default should be 11 db but its not (at least for me?)
> Try setting the attenuation explicitly before analogRead(YOUR_GPIO):
> 
> analogSetPinAttenuation(YOUR_GPIO, adc_attenuation_t::ADC_11db);
> or
> analogSetAttenuation(adc_attenuation_t::ADC_11db);

Using analogSetPinAttenuation() works for me, but analogSetAttenuation() does not. You don't have to use the function every time you do a analogRead(). Just once in setup() is fine, but you have to specify it for each ADC pin that you intend to use. 

I could not set the resolution though. It is fixed at 13 bit (0-8191).

szerwi, 2021-08-08:
@marcboon You're right, ```analogSetPinAttenuation()``` works well, thanks!
Anyway, I hope it will be fixed in 2.0.0-rc2 version ;)

me-no-dev, 2021-08-11:
fixed in master :)

closed 2021-08-11 by me-no-dev