espressif/arduino-esp32 #5478: I2C support

VojtechBartoska, 2021-07-29:
Current state:
- working on ESP32
- doesn't support ESP32-S2 & ESP32-C3 (problem between Write and Read Transaction)
- consider Multithread solution (Thread safe)

To-Do:
Find a solution for I2C support and implement it in next release.

mjs513, 2021-08-01:
Just as another data point was playing with an Adafruit Feather ESP32-S2 in the middle of the night last night using 2.0.0rc1  and ran into this problem using an adafruit OLD 128x64 and a sparkfun ICM90468 Imu. 

Know this is in work but is there any time time line for release with the fix. Understand lots going on - just wondering.

Stresspuppy67, 2021-08-06:
Hi I think  am having the same issue. I am using an ESP32-S2 Saola 1M, with the Aduino IDEI (Mac), using the espressif esp32 2.0.0-rc1 library in the board manager. 

**This is the i2c scanner code I am using:**

#include <Wire.h>
#define I2C_SDA 39  
#define I2C_SCL 40

TwoWire I2CBME = TwoWire(0);

void PrintErrors(bool bResult) {
  return;
  if(!bResult) {
    uint8_t errorNumber = Wire.lastError();
    char * errorText = Wire.getErrorText(errorNumber);
    Serial.print("\nERROR ");
    Serial.print(errorNumber);
    Serial.print(" ");
    if (errorNumber<16) Serial.print("0");
    Serial.println(errorNumber,HEX);
    Serial.print(" ");
    Serial.print(errorText);
  }
}

void PrintError(byte errorNumber) {
  return;
  char * errorText = Wire.getErrorText(errorNumber);
  Serial.print("\nERROR ");
  Serial.print(errorNumber);
  Serial.print(" ");
  if (errorNumber<16) Serial.print("0");
  Serial.println(errorNumber,HEX);
  Serial.print(" ");
  Serial.print(errorText);
}

void setup() {
  //I2CBME.begin(I2C_SDA, I2C_SCL, 100000);
  PrintErrors(Wire.setPins(I2C_SDA, I2C_SCL));
  PrintErrors(Wire.begin());
  Serial.begin(115200);
  Serial.println("\nI2C Scanner");
}
 
void loop() {
  byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  for(address = 1; address < 127; address++ ) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address<16) Serial.print("0");
      Serial.println(address,HEX);
      nDevices++;
    } else {
      Serial.print("Error at address 0x");
      if (address<16) Serial.print("0");
      Serial.println(address,HEX);
      PrintError(error);
      PrintErrors(false);
    }
  }

  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  }
  else {
    Serial.print(nDevices);
    Serial.println(" I2C devices found\n");
  }

  delay(5000);          
}

**This is the error in the Serial Monitor:**

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4002bb93  PS      : 0x00060034  A0      : 0x8002a10f  A1      : 0x3ffc0330  
A2      : 0x3ffcd5f8  A3      : 0x00000001  A4      : 0x00060021  A5      : 0x3ffcd5ac  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffc1fe8  
A10     : 0x00060023  A11     : 0x00000003  A12     : 0x00060023  A13     : 0x3ffcd604  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00060023  LEND    : 0x3ffcd604  LCOUNT  : 0x40026000  
Core  0 was running in ISR context:
EPC1    : 0x40086373  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4002bb93

Backtrace:0x4002bb90:0x3ffc03300x4002a10c:0x3ffc0350 0x40026ee1:0x3ffc0380 0x40027566:0x3ffc03b0 0x4002b51d:0x3ffc7ae0 0x4008959b:0x3ffc7b20 0x40081458:0x3ffc7b70 0x40081265:0x3ffc7b90 0x400812d9:0x3ffc7bb0 0x4008104f:0x3ffc7bd0 0x40081d69:0x3ffc7bf0 

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400264d0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x498
load:0x4004c000,len:0xa74
load:0x40050000,len:0x25ec
entry 0x4004c194

**And this is the output from the exception decoder:**

PC: 0x4002bb93: vPortEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c line 450
EXCVADDR: 0x00000000

Decoding stack results
0x4002bb90: vPortEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c line 450
0x4002a10c: xQueueGenericSendFromISR at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c line 1011
0x40026ee1: i2c_isr_handler_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/i2c.c line 509
0x4002b51d: xTaskGetCurrentTaskHandleForCPU at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c line 4477
0x4008959b: i2c_master_cmd_begin at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/i2c.c line 1477
0x40081458: i2cWrite at /Users/emmyhazan/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/cores/esp32/esp32-hal-i2c.c line 1846
0x40081265: TwoWire::endTransmission(bool) at /Users/emmyhazan/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/libraries/Wire/src/Wire.cpp line 150
0x400812d9: TwoWire::endTransmission() at /Users/emmyhazan/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/libraries/Wire/src/Wire.cpp line 333
0x4008104f: loop() at /Users/emmyhazan/ESP32-S2/I2c_scanner_DEV1/I2c_scanner_DEV1.ino line 54
0x40081d69: loopTask(void*) at /Users/emmyhazan/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-rc1/cores/esp32/main.cpp line 43

I hope someone can help. Many thanks !!!

evalstate, 2021-08-09:
Having the same issue with Unexpected Maker Feather S2; have reverted to alpha1 for the moment

Stresspuppy67, 2021-08-09:
Have managed to get i2c scanner working.

I added the error code for every address for debugging purposes.
If the address has no device attached, it generates error 255.
When the address had a BME280 attached, it generated error 7, and then rebooted.

Turns out the issue is that the example does a Wire.beginTransmission() and Wire.endTransmission()
without doing a Wire.write() in between. The Wire class cleverly stores writes in a buffer, which is initialised with Wire.getTransmission(), stores data in the buffer with Wire.write() and gets sent when the transmission is completed by sending the Wire.endTransmission(). So if you begin a transmission and end it right after, the Wire.endTransmission() is sending an empty array.

Sending an empty array to a non-device, isn't an issue, because there's nothing there.
But when you send an empty array to an existing device, it's expecting data and doesn't get data.
Sending an actual array generates a successful response.

`/*
Author: Roy Hazan
Date: 9th August 2021
*/
// #include "task_watchdog_example_main.c"

#ifdef CORE_DEBUG_LEVEL
#undef CORE_DEBUG_LEVEL
#endif

#define CORE_DEBUG_LEVEL 3
#define LOG_LOCAL_LEVEL ESP_LOG_DEBUG

#include <Wire.h>
#define I2C_SDA 20  
#define I2C_SCL 21

byte address; // For printing error information

TwoWire I2CBME = TwoWire(0);

void PrintErrors(bool bResult) {
  return;
  if(!bResult) {
    uint8_t errorNumber = Wire.lastError();
    char * errorText = Wire.getErrorText(errorNumber);
    Serial.print("\nERROR ");
    Serial.print(errorNumber);
    Serial.print(" ");
    if (errorNumber<16) Serial.print("0");
    Serial.println(errorNumber,HEX);
    Serial.print(" ");
    Serial.print(errorText);
  }
}

void PrintError(byte errorNumber) {
  return;
  char * errorText = Wire.getErrorText(errorNumber);
  Serial.print("\nERROR ");
  Serial.print(errorNumber);
  Serial.print(" ");
  if (errorNumber<16) Serial.print("0");
  Serial.println(errorNumber,HEX);
  Serial.print(" ");
  Serial.print(errorText);
}

void setup() {
  //I2CBME.begin(I2C_SDA, I2C_SCL, 100000);
  PrintErrors(Wire.setPins(I2C_SDA, I2C_SCL));
  PrintErrors(Wire.begin());
  Serial.begin(115200);
  Serial.println("\nI2C Scanner");
}
 
void loop() {
  byte errorByte;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  for(address = 1; address < 127; address++ ) {
    Wire.beginTransmission(address);
    Wire.write("c");
    errorByte = Wire.endTransmission();
    if (errorByte == 0) {
      Serial.print("I2C device found at address 0x");
      if (address<16) Serial.print("0");
      Serial.println(address,HEX);
      nDevices++;
    } else {
      Serial.print("Error ");
      Serial.print(errorByte);
      Serial.print(" = 0x");
      Serial.print(errorByte, HEX);
      Serial.print(" at address 0x");
      if (address<16) Serial.print("0");
      Serial.println(address,HEX);
    }
  }

  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  }
  else {
    Serial.print(nDevices);
    Serial.println(" I2C devices found\n");
  }

  delay(5000);          
}`

janrutger, 2021-08-17:
did not crash, indeed but also did not find my I2C sensor.....

Nasty, it  did work in the Alpha release :(

Stresspuppy67, 2021-08-17:
The code finds the device address. This is the serial output I get, I've highlighted the i2c device.

Default I2C buffer length: 128
Scanning...
Error 255 = 0xFF at address 0x01, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x02, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x03, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x04, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x05, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x06, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x07, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x08, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x09, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x0A, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x0B, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x0C, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x0D, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x0E, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x0F, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x10, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x11, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x12, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x13, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x14, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x15, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x16, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x17, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x18, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x19, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x1A, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x1B, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x1C, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x1D, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x1E, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x1F, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x20, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x21, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x22, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x23, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x24, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x25, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x26, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x27, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x28, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x29, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x2A, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x2B, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x2C, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x2D, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x2E, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x2F, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x30, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x31, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x32, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x33, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x34, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x35, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x36, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x37, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x38, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x39, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x3A, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x3B, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x3C, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x3D, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x3E, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x3F, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x40, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x41, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x42, bus: 1, SDA: 39, SCL: 40
[... 36 lines cut ...]
Error 255 = 0xFF at address 0x67, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x68, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x69, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x6A, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x6B, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x6C, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x6D, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x6E, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x6F, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x70, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x71, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x72, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x73, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x74, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x75, bus: 1, SDA: 39, SCL: 40
**I2C device found at address 0x76, bus: 1, SDA: 39, SCL: 40**
Error 255 = 0xFF at address 0x77, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x78, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x79, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x7A, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x7B, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x7C, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x7D, bus: 1, SDA: 39, SCL: 40
Error 255 = 0xFF at address 0x7E, bus: 1, SDA: 39, SCL: 40
**1 I2C devices found**

janrutger, 2021-08-19:
Hmm, strange I got the **no device found** message. 
But the good news, the published fix for this, to change the ESP_HAL_I2C.C file work for me.
The code is working again as before

SuGlider, 2021-09-14:
@mjs513 @Stresspuppy67  @evalstate  @jandegr 
Please check Arduino Core 2.0.0 with PR #5664 
It may solve the issues you have found.
Let me know. Thanks.

me-no-dev, 2021-09-20:
Hi could you please re-test with this new PR: https://github.com/espressif/arduino-esp32/pull/5683

PR #5683 (2021-10-01): Implement Thread-Safe I2C based on ESP-IDF API
## Summary
This PR is a complete refactoring of  Wire Library and I2C HAL in order to use IDF instead of current Register manipulation approach.

It implements Thread-safe Arduino Wire.

Fix #5478 
Fix #5635 
Fix #5636 
Fix #4729 
Fix #5470 

## Impact
It solves many reported issues related to I2C.
closed 2021-10-01 by me-no-dev