espressif/arduino-esp32 #5481: ESP32 v2.0.0-rc1 not working with HTS221 sensor

badril1978, 2021-07-30:
I am trying to test the latest ESP32 board package with ESP32-S2 board which has an HTS221 and BMP280 sensors attached to it.
When running a simple example to test HTS221 test from the example menu of Arduino studio 1.8.13, after compiling and uploading to the board, I get the following:
"Failed to find HTS221 chip " from the serial monitor.

When I tried using the ESP32-S2 package from the following link:
https://www.mischianti.org/2020/12/01/esp32-s2-pinout-specs-and-arduino-ide-configuration-1/

It worked like a charm. I was able to get readings from HST221 and BMP280 sensors.
Is there something missing in the V2.0.0-rc1 library? Could somebody look into this.

me-no-dev, 2021-07-30:
so with which version did it work? installing master?

badril1978, 2021-08-05:
Sorry for the late reply. I installed from the master following the instructions from the link I referenceed.

VojtechBartoska, 2022-03-30:
Hello @badril1978, sorry for late reply. Are you able to test your issue on development version 2.0.3-RC1. You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release in Arduino IDE.

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska