espressif/arduino-esp32 #5484: often (not always) after Wifi start: Brownout detector was triggered , then repeatedly reboot

dsyleixa, 2021-07-31:
often (not always) after Wifi start: Brownout detector was triggered , then repeatedly reboot:
what can be the reason? 

```

String SSID     = "WLAN-123";
String PASSWORD = "1865-4657";
String StringGW_IP  = "192.168.2.1";
String StringLOC_IP = "192.168.2.202";
String StringSUB_IP = "255.255.255.0";
IPAddress LOCALIP(192, 168,  2, 202);
IPAddress GATEWAY (192, 168,  2,  1);
IPAddress SUBNET  (255, 255, 255, 0);
int WIFIPORT = 8008;
int WEBPORT =  8084;

//...

void(setup) {
   Serial.begin(115200);
   Wire.begin();
   Wire.setClock(400000);
   Adafruit_HX8357_ini(3);
   SD.begin();
   ads.begin(0x48);
   init_MPU6050();   // everything ok until here
   //...
   // debug, to check:
   Serial.println(SSID);
   Serial.println(LOCALIP);
   Serial.println(GATEWAY);
   Serial.println(SUBNET);
   Serial.println(WIFIPORT);
   Serial.println(WEBPORT);  // everything ok until here

   Serial.println("try WiFi ");     // <<<<<<<<<<<<<<< randomly breaks here after this message 
                                   // then Serial message: "Brownout detector was triggered"
                                   //  but sometime also fine, too 
   delay(10);

   WiFi.mode(WIFI_STA);
   WiFi.config(LOCALIP, GATEWAY, SUBNET, GATEWAY, GATEWAY);
   Serial.println(LOCALIP.toString());
   Serial.println(" config: done ");

   WiFi.begin( ssid, pwd );
   Serial.println(" begin: ");
   delay(10);
   // Wait for connection
   while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
   }
   tftline+=15;
   delay(1);
   Serial.println();
   //...               
   //  <<<<<<<<<<<<<<<<   ... but sometimes also randomly fine down to here
}
   
```

dsyleixa, 2021-07-31:
update,
could be improved slightly, to a moderate degree (not comletely though) by hints from this link: 
https://github.com/espressif/arduino-esp32/issues/863#issuecomment-575923300:
WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);

dsyleixa, 2021-07-31:
update,
could be improved slightly, to a moderate degree (not completely though) by hints from these link: 
https://github.com/espressif/arduino-esp32/issues/863#issuecomment-347179737
https://github.com/espressif/arduino-esp32/issues/863#issuecomment-575923300:
(both make actually no big difference)

me-no-dev, 2021-08-01:
brownouts have one reason - not enough current for the esp to boot the radio. Either bad cable or the LDO is not made to handle the current spikes, so it drops the voltage enough to cause the BO

dsyleixa, 2021-08-01:
thank you! So far it always worked, plugged to my PC's USB 3.0 port, and always using the same USB cable.
But so I will try other USB cables and looking forward!

dsyleixa, 2021-08-01:
but out of interest, why does that always happen immediately and reproducible after the command WiFi.mode(WIFI_STA)?

dsyleixa, 2021-08-01:
yes, now I see, that booting  is apparently what WiFi.mode(WIFI_STA) does...

dsyleixa, 2021-08-02:
update:
I found another USB cable by which it works now better, thank you!

closed 2021-08-02 by dsyleixa

josef2600, 2023-07-05:
i found out how to solve it!
https://github.com/josef2600/Brownout-detection-problem-in-Arduino-and-ESP32-C3

dsyleixa, 2023-07-06:
what are the levels for, with which effects (and side-effects)actually, and which one is most suitable?

josef2600, 2023-07-14:
edit 1:
it appears I made few confusions! First, ESP32 and ESP32C3 are way different from each other. C3 has a very very high voltage trigger. E.g., Esp32 lowest trigger level is around 2.44 volt, but C3 is over 3 volts. So that is why C3 has a big problem.
Second:
CONFIG_ESP32C3_BROWNOUT_DET_LVL 7 is the highest level, meaning, the highest voltage needed. I think it is 3.2-volt trigger. It is a very happy trigger chip! And CONFIG_ESP32C3_BROWNOUT_DET_LVL 1 is around 3 (a bit more than 3) volt. If you use usb or 5 volts for input and give it to a low drop regulator, it is high possibility to reset the chip. If you want to be sure, do this:
Enable debug level info. Then try to connect to it via WIFI or do something that causes reset. Then when it resets, it prints in serial port, what caused the reset. If it is brownout reset, it prints it.

dsyleixa, 2023-07-14:
I have it attached by an  USB cable connected to my Windows PC - USB port, running the Arduino IDE, and the Serial monitor window is opened, so I can see. the message that the "brownout" detector was triggered and thus the ESP32 is rebooting.

josef2600, 2023-07-16:
as i said, it is caused by low pick voltage. if you need to use the 5 volt, you have to use a special smps to convert 5 to 3.3 volt. but, only for testing, you can do this to your code to disable BROWN_OUT  detector:

#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"

void setup(void)
{
    WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable BROWN_OUT  detector
...
}

josef2600, 2023-07-16:
https://github.com/josef2600/Brownout-detection-problem-in-Arduino-and-ESP32-C3/blob/main/disable%20BROWN_OUT%20%20detector

dsyleixa, 2023-07-16:
what is a smps?
OTOH, I do not need to "convert" 5v to 3v3 because this is done by my ESP32 Feather USB adapter.

dsyleixa, 2023-07-16:
what is a smps?
OTOH, I do not need to "convert" 5v to 3v3 because this is done by my ESP32 Feather USB adapter.

Additionally I once had already disabled the brownout detector, but that does not actually solve the underlying issue of voltage:
The ESP2 is simply expected to always work fine with either standard voltage supply, but as already stated:
I found another USB cable by which it works now better.

josef2600, 2023-07-17:
SMPS = Switched Mode Power Supply.
eh! it gets a little hard for me! i always assume every body knows everything ! i do not use a made board. i design and build my own board. but i do mistakes too, always! anyway, on that board you are using, it is using a regulator ic AMS1117-3.3 to change the voltage from 5 volt to 3.3 volt. but it causes 2 major problems.
1. AMS1117-3.3 is always a fake chip. i have never seen an original chip on a cheap board. and i haven't fined an original chip in more than 3 years, because it is used in so many big products like computer parts and there is a very bad shortage of original chip.
2. base on datasheet, the minimum Dropout Voltage is 1.250 volt. but we don't have the original chip and we have a dropout voltage on the wires on USB too. the chip consumes up to 500 ma and it is alot! it is piked, meaning, it is 100ma and when it tries to turn WiFi on, it consumes 500 ma in a short time. 500 ma causes voltage drop on cables and suddenly, instead of 5 volt, we get 4 volt or less. and the output drops too. and the chip resets at 3 volt. 

so what can we do? use more voltage (less than 10 volt, the ams1117-3.3 max out at 12 volt and shorts the input to output!, strong cables, or best of all, use a switching regulator. i haven't mention any number, because i haven't finished my tests. but you can use TPS54302HDDCT or TPS54202HDDCT for it, if you can design it. remember if you do want to choose those, choose the self for 10 UH to 8 UH and not more.
hope this time it was enough info.

dsyleixa, 2023-07-17:
thanks, but I cannot use more voltage because this is provided by the USB port of my Windows Notebook  or a "power bank" and I also can't change anything on my Adafruit ESP32 Feather PCB.