espressif/arduino-esp32 #5485: Compilation of NimBLE-based ble keyboard program into ESP32 C3 error.

Danjuanlab, 2021-07-31:
### Hardware:
Board:							ESP32 C3 chip revision: 3，
Core Installation version:			2.0.0 RC1
IDE name:						Arduino IDE? 
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I used the ble keyboard sample code based on the NimBLE library to write code to the esp32 c3 development board, but the compilation error occurred. From the code point of view, the C3 config file lacks some variable definitions and type methods. I don't know if this is a problem with the NimBLE library or the lack of information in the ESP32 c3 library.

NimBLE：https://github.com/h2zero/NimBLE-Arduino

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```#include <OneButton.h>
#include <BleKeyboard.h>

BleKeyboard bleKeyboard;
OneButton b1(1,true);

void setup() {
  
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  b1.attachClick(sev);
  bleKeyboard.begin();
}

void loop() {
  if(bleKeyboard.isConnected()) {
    b1.tick();
    Serial.println("Sending 'Hello world'...");
  }

  Serial.println("Waiting 5 seconds...");
  delay(5000);
}

void sev(){
    Serial.println("Sending Ctrl+Alt+Delete...");
    bleKeyboard.press(KEY_F8);
    delay(100);
    bleKeyboard.releaseAll();
  }
```

### Debug Messages:
```
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:65:65: error: 'CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE' was not declared in this scope
 uint16_t                    NimBLEDevice::m_scanDuplicateSize = CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE;
                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:65:65: note: suggested alternative: 'CONFIG_BT_CTRL_SCAN_DUPL_CACHE_SIZE'
 uint16_t                    NimBLEDevice::m_scanDuplicateSize = CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE;
                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                 CONFIG_BT_CTRL_SCAN_DUPL_CACHE_SIZE
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:66:62: error: 'CONFIG_BTDM_SCAN_DUPL_TYPE' was not declared in this scope
 uint8_t                     NimBLEDevice::m_scanFilterMode = CONFIG_BTDM_SCAN_DUPL_TYPE;
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:66:62: note: suggested alternative: 'CONFIG_BT_CTRL_SCAN_DUPL_TYPE'
 uint8_t                     NimBLEDevice::m_scanFilterMode = CONFIG_BTDM_SCAN_DUPL_TYPE;
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                                              CONFIG_BT_CTRL_SCAN_DUPL_TYPE
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp: In static member function 'static void NimBLEDevice::init(const string&)':
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:554:16: error: 'struct esp_bt_controller_config_t' has no member named 'mode'
         bt_cfg.mode = ESP_BT_MODE_BLE;
                ^~~~
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:555:16: error: 'struct esp_bt_controller_config_t' has no member named 'ble_max_conn'; did you mean 'ble_max_act'?
         bt_cfg.ble_max_conn = CONFIG_BT_NIMBLE_MAX_CONNECTIONS;
                ^~~~~~~~~~~~
                ble_max_act

```

lbernstone, 2021-07-31:
Did you try using the esp32c3 branch of the NimBLE library which looks like it is specifically written to get around these issues?
https://github.com/h2zero/NimBLE-Arduino/tree/esp32c3

Danjuanlab, 2021-07-31:
@lbernstone   Thank you, It is work~

closed 2021-07-31 by Danjuanlab