espressif/arduino-esp32 #5489: Cannot use the latest 'framework-arduinoespressif32' on pio

ullix, 2021-08-02:
I tried to use the latest `framework-arduinoespressif32` platform_packages for ESP32 on pio, but all failed. Something is missing, where can I get it?

I use this pio.ini setting:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

platform_packages =
 ;framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#1.0.6	 :#1
 ;framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32              ;#2
 ;framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#2.0.0-rc1    ;#3
 ;framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#2.0.0-alpha1 ;#4
```
and then un-comment one of platform_packages lines, marked 1 ... 4 (at end of line).

Using line 1 seems to give the same as not selecting any of the platform-packages at all, and it compiles and works fine. Lines 2, 3, 4 all give the same result, namely compiling fails at linking with:
```
Linking .pio/build/esp32dev/firmware.elf
~/.platformio/packages/toolchain-xtensa32@2.50200.97/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o:
~/.platformio/packages/toolchain-xtensa32@2.50200.97/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o:
```

So, crt1-sim.o and  _vectors.o are missing. Where do I find them?

Is that actually all that is missing, or is it rather the sign of an even bigger problem?

lbernstone, 2021-08-02:
https://github.com/espressif/arduino-esp32/issues/5436#issuecomment-885469600

ullix, 2021-08-02:
no luck so far. using this:
```
[env:esp32dev]
; platform = espressif32
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#2.0.0-rc1
```

compiles ok, but then gives this:
```
Linking .pio/build/esp32dev/firmware.elf
~/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/lib4a8/libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x8): undefined reference to `mbedtls_md5_starts'
~/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/lib4a8/libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':
~/Code/platformio/ampp-v0.1/.pio/libdeps/esp32dev/ESP Async WebServer/src/WebAuthentication.cpp:74: undefined reference to `mbedtls_md5_starts'
```

lbernstone, 2021-08-02:
I think you missed this part:

> This workflow is temporary and once the core v2.0 is officially released, everything will get back to normal. In case you still have any problem, please report them directly to platformio/platform-espressif32

me-no-dev, 2021-08-02:
aaand you need to update your Async libs

ullix, 2021-08-03:
Pio pretends there is nothing to upgrade. Unless you can provide some guidance, there is nothing I can do any more :-(

Jason2866, 2021-08-04:
You can tell pio which lib and version you want to use. Time to read pio docu ;-)

jwchen119, 2021-09-05:
> Pio pretends there is nothing to upgrade. Unless you can provide some guidance, there is nothing I can do any more :-(

Instead of using `ESP Async WebServer`, use `https://github.com/me-no-dev/ESPAsyncWebServer.git`

ullix, 2021-09-06:
@jwchen119 Thank you. In the meantime I found help elsewhere https://esp32.com/viewtopic.php?f=19&t=22703&sid=961001ed60ba45092f1c038c8ec17cdb#p81841
Note that one also has to use a different version of AsyncTCP!

VojtechBartoska, 2022-03-30:
Hello, I'm closing this issue you can found and link issues in #6044 for PIO setup.

closed 2022-03-30 by VojtechBartoska